mantidproject/mantid #32918: Detector ID to Workspace Index Mapping Broken in KafkaEventStreamDecoder for come ISIS Instruments

LamarMoore, 2021-11-08:
**Original reporter:** Lamar Moore (ISIS)

**Describe the bug**
The `KakfaEventStreamDecoder` makes use of the spectrum number to workspace index mapping to populate workspaces with events. A shortcut to increase speed making use of `MatrixWorkspace::getSpectruumToWorkspaceIndexVector` which is only useful where spectrum numbers are entirely contiguous. Otherwise this returns garbage silently, one should use the alternative, and less performant `MatrixWorkspace::getSpectrumToWorkspaceIndexMap()` where these numbers are not contiguous.

**To Reproduce**
This can be reproduced by updating the facilities file to include the following modifications for MAPS and running the LiveData streamer from the GUI. Setting the run to begin at the `Start of Run` as opposed to `Now`:
```xml
<instrument name="MAPS" shortname="MAP">
    <technique>Neutron Spectroscopy</technique>
    <technique>TOF Direct Geometry Spectroscopy</technique>
    <livedata default="event">
      <connection name="event" address="itachi.isis.cclrc.ac.uk:9092" listener="KafkaEventListener" />
	  <topic name="MAPSTEST_sampleEnv" type="sample" />
	  <topic name="MAPSTEST_runInfo" type="run" />
	  <topic name="MAPSTEST_events" type="event" />
	  <topic name="MAPSTEST_monitorHistograms" type="monitor" />
    </livedata>
  </instrument>
```

**Expected behavior**
This will cause Mantid to crash due with an array out of bounds exception.

**Platform/Version (please complete the following information):**
Should affect all platforms

**Additional context**
This is needed by members of the technology department (contact: Jack Day) in order to complete some instrumentation tests on the MAPS instrument.

martyngigg, 2021-11-09:
In additiona to the issue above I think there might be another issue here too.

At ISIS the mapping of detectors to spectra is not an intrinsic part of the geometry but rather provided by electronics and can in theory change per run. This mapping has traditionally been read from ISIS RAW or NeXus files and stored in the workspace.

The [kafka event decoder](https://github.com/mantidproject/mantid/blob/main/Framework/LiveData/src/Kafka/KafkaEventStreamDecoder.cpp#L536) already has a check whether the stream contains a detector mapping field. If so the workspace is created using this mapping with the appropriate number of event lists for the number of spectra. As far as I can tell from the MAPS Kafka stream no such mapping is specified, i.e. [`runStartData.detSpecMapSpecified`](https://github.com/mantidproject/mantid/blob/3ba8b76abc132476034e8d2932e3d88e2b69b9e7/Framework/LiveData/src/Kafka/KafkaEventStreamDecoder.cpp#L536) returns `false` and the first branch is taken. Without the mapping from the stream there is no way that the spectra-detector mapping will be correct even fixing the original issue noted and going via the slower MatrixWorkspace map methods.

martyngigg, 2021-11-09:
I also think a naming issue exists in the code. The event stream returns detector IDs but the code is currently suggesting it maps from spectrum numbers back to workspace indices. This is likely only a problem at ISIS as other facilities treat these numbers as the same thing.

closed 2021-12-15 by DannyHindson