mantidproject/mantid #32987: tables in groups tab of muon nalysis are too small

Harrietbrown, 2021-11-16:
In muon analysis the boxes for the tables are too small, users can only see one at a time in regular resolution

**Original reporter:** Aidrian Hillier

sf1919, 2021-11-16:
Further information - AH is most concerned about the first box on the tab showing more than 1 line as often not using the other boxes.

AnthonyLim23, 2021-11-16:
can we have a screenshot please?

sf1919, 2021-11-16:
[image]

AnthonyLim23, 2021-11-17:
I thought this might happen when we added the difference table and we did warn them. Without increasing the minimum height or removing stuff it is going to be difficult. 

The only potential thing I can see is the group range being on one line instead of 2. However, I think the other 2 tables will still be a single row.

closed 2022-06-04 by stale[bot]