mantidproject/mantid #33018: DeleteWorkspace makes mantid hang for WISH data after NormaliseToMonitor (windows not IDAaaS)

RichardWaiteSTFC, 2021-11-22:
**Describe the bug**
Windows hangs when deleteing a WISH workspace above a certain memory threshold.

**To Reproduce**
(1) Run this script
```
LoadRaw(Filename='WISH00043526.raw', OutputWorkspace='Vana', LoadMonitors='Separate') # 2021-11-10T16:30:01.487334000 execCount: 19
CropWorkspace(InputWorkspace='Vana', OutputWorkspace='Vana', XMin=6000, XMax=99000) # 2021-11-10T16:30:31.007140000 execCount: 20
NormaliseByCurrent(InputWorkspace='Vana', OutputWorkspace='Vana') # 2021-11-10T16:30:54.491781000 execCount: 21
ConvertUnits(InputWorkspace='Vana', OutputWorkspace='Vana', Target='Wavelength') # 2021-11-10T16:30:55.211130000 execCount: 22
NormaliseByCurrent(InputWorkspace='Vana_monitors', OutputWorkspace='Vana_monitors') # 2021-11-10T16:30:55.527786000 execCount: 23
CropWorkspace(InputWorkspace='Vana_monitors', OutputWorkspace='Vana_monitors', XMin=6000, XMax=99000) # 2021-11-10T16:30:55.532764000 execCount: 24
ConvertUnits(InputWorkspace='Vana_monitors', OutputWorkspace='Vana_monitors', Target='Wavelength') # 2021-11-10T16:30:55.537742000 execCount: 25
NormaliseToMonitor(InputWorkspace='Vana', OutputWorkspace='Vana_norm', MonitorWorkspace='Vana_monitors', MonitorWorkspaceIndex=3) # 2021-11-10T16:30:55.541725000 execCount: 26
```
(2) Delete the `Vana` workspace - it hangs on windows 

(3) Load in a small section of WISH data (1 bank only) - it will no longer hang
```
LoadRaw(Filename='WISH00043526.raw', OutputWorkspace='Vana', LoadMonitors='Separate',
    SpectrumMin=1, SpectrumMax=19461)  # bank 1
```

**Expected behavior**
It wouldn't hange and tghe delete workspace would execute successfully in ~5s

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: 6.2 and nightly

**Additional context**

RichardWaiteSTFC, 2022-03-03:
@thomashampson have core got any capacity to look into this next sprint? WISH have mentioned it again.

sf1919, 2024-02-27:
I've been unable to recreate this on v6.9 on Windows. The workspace deletes quickly with no hanging. Could @RichardWaiteSTFC and @thomashampson also check whether it's reproducable still?

sf1919, 2024-02-27:
I've just tried deleting `Vana_norm` and that is hanging so there is definitely some issue still there

RichardWaiteSTFC, 2024-02-27:
> I've just tried deleting `Vana_norm` and that is hanging so there is definitely some issue still there

I've also had WISH scientists report the issue when deleting `Vana_norm`

sf1919, 2024-07-09:
@RichardWaiteSTFC can we retry this on v6.10? I'm not sure I can recreate it so it might have been fixed?