mantidproject/mantid #33027: FrameworkManager Shutdown Races Qt threading (via Poco Notifications)

DavidFair, 2021-11-22:
Notice as part of #32942 , also I've personally worked around this multiple times

**Describe the bug**
- Currently the FrameworkImpl will `clearData` whilst running through shutdown
- At the same time Qt will be shutting down
- As part of the ADS shutdown it will post a [clear notification](https://github.com/mantidproject/mantid/blob/9acdc0e188dc5d38a4fbee109b218b21d532892e/Framework/Kernel/inc/MantidKernel/DataService.h#L340)
- Poco dutifully forwards this to an observing widget
- In this case `WorkspaceSelect::handleClearEvent` will try to invoke [clear](https://github.com/mantidproject/mantid/blob/9acdc0e188dc5d38a4fbee109b218b21d532892e/qt/widgets/common/src/WorkspaceSelector.cpp#L215) on the underlying QObject 
- As part of the Qt clean code it will try to invoke `QtGui!QGUIApplication::font()`, but this has been unloaded
- Segmentation fault

For reference the full backtrace is:
```
  * frame #0: 0x000000011102c61d QtGui`QGuiApplication::font() + 141
    frame #1: 0x00000001110cb134 QtGui`QFont::QFont() + 36
    frame #2: 0x0000000110ac8187 QtWidgets`QStyleOptionComboBox::QStyleOptionComboBox() + 71
    frame #3: 0x0000000110b4ef49 QtWidgets`___lldb_unnamed_function1210$$QtWidgets + 57
    frame #4: 0x0000000110b52df7 QtWidgets`___lldb_unnamed_function1229$$QtWidgets + 471
    frame #5: 0x0000000111802c35 QtCore`___lldb_unnamed_function2616$$QtCore + 1477
    frame #6: 0x0000000111781360 QtCore`QAbstractItemModel::endRemoveRows() + 272
    frame #7: 0x00000001113a6506 QtGui`QStandardItemModel::removeRows(int, int, QModelIndex const&) + 86
    frame #8: 0x0000000110b57e3e QtWidgets`QComboBox::clear() + 94
    frame #9: 0x0000000106d5f2b3 libMantidQtWidgetsCommonQt5.dylib`MantidQt::MantidWidgets::WorkspaceSelector::handleClearEvent(this=0x0000000160778f40, (null)=0x00007ffeefbfe788) + 99 at WorkspaceSelector.cpp:215
    frame #10: 0x0000000106d6300f libMantidQtWidgetsCommonQt5.dylib`Poco::NObserver<MantidQt::MantidWidgets::WorkspaceSelector, Mantid::Kernel::DataService<Mantid::API::Workspace>::ClearNotification>::notify(this=0x00000001607808e0, pNf=0x00000001470e2170) const + 303 at NObserver.h:86
    frame #11: 0x0000000101c73363 libPocoFoundation.71.dylib`Poco::NotificationCenter::postNotification(Poco::AutoPtr<Poco::Notification>) + 141
    frame #12: 0x00000001000c9f1a MantidScientificInterfacesISISReflectometryTestQt5`Mantid::Kernel::DataService<Mantid::API::Workspace>::clear(this=0x000000015af32e80) + 234 at DataService.h:340
    frame #13: 0x000000011353edb9 libMantidAPI.dylib`Mantid::API::FrameworkManagerImpl::clearData(this=0x000000011c02fc00) + 25 at FrameworkManager.cpp:234
    frame #14: 0x000000011353ed4b libMantidAPI.dylib`Mantid::API::FrameworkManagerImpl::clear(this=0x000000011c02fc00) + 43 at FrameworkManager.cpp:215
    frame #15: 0x000000011353eeaa libMantidAPI.dylib`Mantid::API::FrameworkManagerImpl::shutdown(this=0x000000011c02fc00) + 202 at FrameworkManager.cpp:223
    frame #16: 0x0000000143009fd2 _api.so`_object* boost::python::detail::invoke<int, void (Mantid::API::FrameworkManagerImpl::*)(), boost::python::arg_from_python<Mantid::API::FrameworkManagerImpl&> >((null)=invoke_tag_<true, true> @ 0x00007ffeefbfe9e8, (null)=0x00007ffeefbfea1c, f=0x0000000138f80828, tc=0x00007ffeefbfea40)(), boost::python::arg_from_python<Mantid::API::FrameworkManagerImpl&>&) + 114 at invoke.hpp:92
    frame #17: 0x0000000143009ed9 _api.so`boost::python::detail::caller_arity<1u>::impl<void (this=0x0000000138f80828, args_=0x0000000166827df0, (null)=0x0000000000000000)(), boost::python::default_call_policies, boost::mpl::vector2<void, Mantid::API::FrameworkManagerImpl&> >::operator()(_object*, _object*) + 185 at caller.hpp:233
    frame #18: 0x0000000143009dc4 _api.so`boost::python::objects::caller_py_function_impl<boost::python::detail::caller<void (Mantid::API::FrameworkManagerImpl::*)(), boost::python::default_call_policies, boost::mpl::vector2<void, Mantid::API::FrameworkManagerImpl&> > >::operator(this=0x0000000138f80820, args=0x0000000166827df0, kw=0x0000000000000000)(_object*, _object*) + 52 at py_function.hpp:38
    frame #19: 0x0000000107a91212 libboost_python38-mt.dylib`boost::python::objects::function::call(_object*, _object*) const + 690
    frame #20: 0x0000000107a93aea libboost_python38-mt.dylib`boost::detail::function::void_function_ref_invoker0<boost::python::objects::(anonymous namespace)::bind_return, void>::invoke(boost::detail::function::function_buffer&) + 26
    frame #21: 0x0000000107a9a01a libboost_python38-mt.dylib`boost::python::detail::exception_handler::operator()(boost::function0<void> const&) const + 74
    frame #22: 0x0000000142f45375 _api.so`boost::python::detail::translate_exception<Mantid::API::Algorithm::CancelException, void (*)(Mantid::API::Algorithm::CancelException const&)>::operator(this=0x0000000138f66928, handler=0x0000000138f66920, f=0x00007ffeefbff0e8, translate=(_api.so`(anonymous namespace)::translateCancel(Mantid::API::Algorithm::CancelException const&) at Algorithm.cpp:98))(boost::python::detail::exception_handler const&, boost::function0<void> const&, void (*)(Mantid::API::Algorithm::CancelException const&)) const + 37 at translate_exception.hpp:46
    frame #23: 0x0000000142f45321 _api.so`bool boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(Mantid::API::Algorithm::CancelException const&)> >::operator(this=0x0000000138f66930, (null)=type<bool> @ 0x00007ffeefbfec38, f=0x0000000138f66928, a=0x00007ffeefbfec68, (null)=0)<bool, boost::python::detail::translate_exception<Mantid::API::Algorithm::CancelException, void (*)(Mantid::API::Algorithm::CancelException const&)>, boost::_bi::rrlist2<boost::python::detail::exception_handler const&, boost::function0<void> const&> >(boost::_bi::type<bool>, boost::python::detail::translate_exception<Mantid::API::Algorithm::CancelException, void (*)(Mantid::API::Algorithm::CancelException const&)>&, boost::_bi::rrlist2<boost::python::detail::exception_handler const&, boost::function0<void> const&>&, long) + 129 at bind.hpp:388
    frame #24: 0x0000000142f45295 _api.so`bool boost::_bi::bind_t<bool, boost::python::detail::translate_exception<Mantid::API::Algorithm::CancelException, void (*)(Mantid::API::Algorithm::CancelException const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(Mantid::API::Algorithm::CancelException const&)> > >::operator(this=0x0000000138f66928, a1=0x0000000138f66920, a2=0x00007ffeefbff0e8)<boost::python::detail::exception_handler const&, boost::function0<void> const&>(boost::python::detail::exception_handler const&&&, boost::function0<void> const&&&) + 85 at bind.hpp:1318
    frame #25: 0x0000000142f44eed _api.so`boost::detail::function::function_obj_invoker2<boost::_bi::bind_t<bool, boost::python::detail::translate_exception<Mantid::API::Algorithm::CancelException, void (*)(Mantid::API::Algorithm::CancelException const&)>, boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(Mantid::API::Algorithm::CancelException const&)> > >, bool, boost::python::detail::exception_handler const&, boost::function0<void> const&>::invoke(function_obj_ptr=0x0000000138f66928, a0=0x0000000138f66920, a1=0x00007ffeefbff0e8) + 45 at function_template.hpp:137
    frame #26: 0x0000000142f45375 _api.so`boost::python::detail::translate_exception<Mantid::API::Algorithm::CancelException, void (*)(Mantid::API::Algorithm::CancelException const&)>::operator(this=0x0000000138f63828, handler=0x0000000138f63820, f=0x00007ffeefbff0e8, translate=(_api.so`(anonymous namespace)::translateCancel(Mantid::API::Algorithm::CancelException const&) at Algorithm.cpp:98))(boost::python::detail::exception_handler const&, boost::function0<void> const&, void (*)(Mantid::API::Algorithm::CancelException const&)) const + 37 at translate_exception.hpp:46
    frame #27: 0x0000000142f45321 _api.so`bool boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::_bi::value<void (*)(Mantid::API::Algorithm::CancelException const&)> >::operator(this=0x0000000138f63830, (null)=type<bool> @ 0x00007ffeefbfed78, f=0x0000000138f63828, a=0x00007ffeefbfeda8, (null)=0)<bool, boost::python::detail::translate_exception<Mantid::API::Algorithm::CancelException, void (*)(Mantid::API::Algorithm::CancelException const&)>, boost::_bi::rrlist2<boost::python::detail::exception_handler const&, boost::function0<void> const&> >(boost::_bi::type<bool>, boost::python::detail::translate_exception<Mantid::API::Algorithm::CancelException, void [... cut ...]
We could write a system to make sure that everything tears down in the correct order, or better still when we call FrameworkImpl::shutdown we know the "demolition team" is behind us. Skip cleanup and let the various destructors tidy everything up as they exit scope:
- Modify `FrameworkManagerImpl::Shutdown()` to call a dedicated `ADS::Shutdown` method
- Protect `ADS::Shutdown` using a key pattern in the method arguments, as we don't want people using this if we're not shutting down...
- Inside ADS::Shutdown we disable the Poco Notification centre before calling clear, to avoid pointless work being done

martyngigg, 2021-11-23:
I guess the key hear is ensuring the notifications are all diconnected - probably more than just the ADS. Things like ConfigService too?

We're currently using Python `atexit` to try and call FrameworkManager shutdown in a timely manner. Perhaps tat just needs to not call `clearData` and just disconnect everything?

DavidFair, 2021-11-23:
> I guess the key hear is ensuring the notifications are all diconnected - probably more than just the ADS. Things like ConfigService too?
>
> We're currently using Python `atexit` to try and call [...]

I agree completely, I'd want to protect a `Shutdown` method which simply disconnects without clearing and let the destructor go through it all.
I agree and it should  ultimately be the same for the ADS and ConfigManager (but it's a good shout to disconnect notifications on all of them)

gemmaguest, 2022-03-08:
Closing as stale for now because we are unlikely to get round to this

closed 2022-03-08 by gemmaguest