mantidproject/mantid #33030: Paalman Pings for Polaris Total Scattering

DanielMurphy22, 2021-11-23:
**Is your feature request related to a problem? Please describe.**
The total scattering workflow for POLARIS currently produces and incorrect Pair Distribution Function because the Placzek correction is larger than the measured (and can corrected) intensity. 

**Describe the solution you'd like**
This may be due to 'over-subtracting' the empty can. To resolve this, the PaalmanPings calculation should be made before subtracting the empty can.

**Additional context**
Probably want to use https://docs.mantidproject.org/nightly/algorithms/PaalmanPingsAbsorptionCorrection-v1.html but worth checking which of the Mantid Paalman Pings algorithms should be used?

SpencerHowells, 2021-11-23:
The best version is  PaalmanPings MonteCarlo. An interface to run it is available in Indirect>Corrections. Danny is the best person to contact for info for diffraction.

DanielMurphy22, 2022-02-09:
This oversubtracting actually applies to all the powder instruments, so might be something we could extend to them all at some stage.

DanielMurphy22, 2022-02-09:
Adding this info here as I'm separating the branches for placezk (https://github.com/mantidproject/mantid/pull/33008) and paalmanpings [image]

DanielMurphy22, 2022-03-03:
**Context and remaining work**

The proof of concept branch with hard coded inputs is here: https://github.com/mantidproject/mantid/pull/33538

Remaining:
- Get input parameters for paalman pings
- This requires including container details in sample details
- add a switch which is defaulted to off as this can take a long time to run compared to the orginial workflow.

closed 2022-07-01 by gemmaguest