mantidproject/mantid #33044: Workbench closes when changing tab on Muon Analysis MacOS

DanielMurphy22, 2021-11-23:
**Describe the bug**
Workbench closes when changing tab in the Muon Analysis interface on MacOS.

**To Reproduce**
- Open MantidWorkbench on MacOS
- Open Muon Analysis or FDA interfaces
- change tab, e.g. to the Grouping tab.
- Workbench closes and an error reporter with no stacktrace appears.
- A MacOS system error report said that it failed with: 
``` 
1   libQt5Gui.5.12.9.dylib        	0x0000000169f6a1ba 
QScreen::grabWindow(unsigned long long, int, int, int, int) + 368
```

**Expected behavior**
The expected tab should open. If there is a problem, Workbench should still be able to stay open.

**Platform/Version (please complete the following information):**
 - OS Version: MacOS BigSur
 - Mantid Version Found on recent nigthly, but also exists in release 6.2 and 6.0

**Additional context**
This meant I could not test this PR on MacOS as I could not get to the Sequential Fitting tab: https://github.com/mantidproject/mantid/pull/33010

I suspect this issue may be lower priority than currently planned work as this has been around for a few releases and the current user base doesn't seem to use MacOS, but these errors could be something we are luckily getting away with on other platforms.

AnthonyLim23, 2021-11-23:
@DanielMurphy22 do you know who has a mac and could work on this? I would like to get it done ASAP, as external users often have macs.

AnthonyLim23, 2021-11-24:
I am happy to help whoever works on this.

AnthonyLim23, 2021-11-24:
It is probably either https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqtinterfaces/mantidqtinterfaces/Muon/GUI/Common/dock/dockable_tabs.py or https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqtinterfaces/mantidqtinterfaces/Muon/GUI/MuonAnalysis/muon_analysis_2.py#L217

PR #33141 (2022-01-03): Fix Crash When Changing Tabs on MacOS
**Description of work.**
Removed some unused code that was causing crashes when on MacOS with multiple monitors. 

`GrabWindow` has a bug in Qt on MacOS where multiple monitors can have a divide by 0 error and crash. 

**To test:**
1. Check that the GUI still works and the normal workflow is unchanged on other OSs (particularly windows)
2. (*MacOS only*): Open the Muon Analysis and FDA GUIs and check that all the tabs can be switched to by single clicking and detatched by double clicking

Fixes #33044

---

#### Reviewer ####

Please comment on the followi [... cut]
closed 2022-01-03 by gvardany