mantidproject/mantid #33048: Scope / Design GSASII integration

DanielMurphy22, 2021-11-23:
Should be part of the Epic: https://github.com/mantidproject/mantid/issues/33041

**Describe the outcome that is desired.**
Scope and start to design the implementation for integrating GSAS-II with the Engineering Diffraction interface.

**Describe any solutions you are considering**
Need to gather the requirements from the last meeting with Saurabh and looking at the features in the old GSASIIRefineFitPeaks algorithm. Think of the potential ways the interface could look and produce a design document. A particular sticking point is how to read back in the results from GSAS-II and report them in Workbench.

DannyHindson, 2022-02-07:
Dan's notes:
https://stfc365-my.sharepoint.com/:w:/g/personal/daniel_murphy_stfc_ac_uk/EY3AjRL20bdHuKqM7sq_t6IBowRAmFSb269YrmQGALIaag?e=GaKijz

DanielMurphy22, 2022-02-24:
Developments are on branch: [33048_GSAS_scripting](/mantidproject/mantid/tree/33048_GSAS_scripting)

DannyHindson, 2022-03-10:
Good stuff! This is progressing nicely. Did the code in the original python algorithm use the "Generated Pawley reflections" approach?

RichardWaiteSTFC, 2022-03-14:
Mantid can generate reflections if provided with lattice parameters and spacegroup like so
```
from mantid.geometry import CrystalStructure, ReflectionGenerator, ReflectionConditionFilter
silicon = CrystalStructure("5.431 5.431 5.431", "F d -3 m", "Si 0 0 0 1.0 0.05")
generator = ReflectionGenerator(silicon)
# Create list of unique allowed reflections between 0.7 and 3.0 Angstrom
hkls = generator.getUniqueHKLsUsingFilter(0.7, 3.0, ReflectionConditionFilter.StructureFactor)
dValues = generator.getDValues(hkls)
```
but GSAS will be able to do this for each phase, and I think it would be better that way

DanielMurphy22, 2022-06-15:
Separate to the sprint planning we should decide how much of the scope / design I have completed and probably write another detailed issue for implementing this!

closed 2022-06-22 by DannyHindson