mantidproject/mantid #33061: Progress new Mantid webpage

DanielMurphy22, 2021-11-25:
**Describe the outcome that is desired.**
A new mantid webpage has been designed and started. It will replace the current Mantid wiki pages but not the user or dev docs. This issue is to progress and hopefully roll out the new website!

**Describe any solutions you are considering**
- See the open PR and issues here: https://github.com/mantidproject/www.mantidproject.org
- Discuss with @martyngigg before starting!

martyngigg, 2022-05-03:
Discussing with the TWG we agreed we'd like to release this with 6.4

martyngigg, 2022-06-13:
A preview is at https://github.io/mantidproject/www.mantidproject.org

DanielMurphy22, 2022-06-13:
A review of the preview: 
- It would be nice if we could detect darkmode and use: https://github.com/mantidproject/mantid/blob/main/images/Mantid_Logo_Transparent_for_dark_backgrounds.png which we already have. Or a quicker solution is to always use a logo with a white background which I've made here:
[image]
- Can we align the title names to the pages they link to? :
Develop -> Developer Documentation :white_check_mark: 
(also good Forums, Release Notes, Contact Us) :white_check_mark: 
User Docs -> Mantid :x: (_rename mantid docs page to User Docs_)
Tutorials -> Training :x: (_rename Training page in mantid docs to Tutorials_)
Downloads -> Installation :x: (_rename new internal page to Downloads_)

martyngigg, 2022-10-26:
The new page is now live at https://mantidproject.org.

closed 2022-10-26 by martyngigg