mantidproject/mantid #33118: Mantid doesn't work with latest setuptools on Windows

DannyHindson, 2021-12-10:
I recently updated the Python setuptools package on my Windows laptop by accident and I noticed that workbench and system tests stopped working. I looked into this a bit and came across some issues where support had been added to Mantid for setuptools v49.0.0 or later (eg https://github.com/mantidproject/mantid/pull/32068).

The problem on Windows seems to be that the python version is 3.8 which includes a new importlib.metadata module. This alters the behaviour of the shim (workbench-script.pyw) created by setuptools v49.0.0 or later:

```
#!C:\mantid\external\src\ThirdParty\lib\python3.8\pythonw.exe
# EASY-INSTALL-ENTRY-SCRIPT: 'MantidWorkbench','gui_scripts','workbench'
import re
import sys

# for compatibility with easy_install; see #2198
__requires__ = 'MantidWorkbench'

try:
    from importlib.metadata import distribution
except ImportError:
    try:
        from importlib_metadata import distribution
    except ImportError:
        from pkg_resources import load_entry_point

def importlib_load_entry_point(spec, group, name):
    dist_name, _, _ = spec.partition('==')
    matches = (
        entry_point
        for entry_point in distribution(dist_name).entry_points
        if entry_point.group == group and entry_point.name == name
    )
    return next(matches).load()

globals().setdefault('load_entry_point', importlib_load_entry_point)

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(load_entry_point('MantidWorkbench', 'gui_scripts', 'workbench')())

```

Ubuntu uses python 3.6 so the shim imports pkg_resources instead.

This difference matters because it appears to me that importlib doesn't recognise .egg-link files whereas pkg_resources does.

Mantid works OK on older versions of setuptools because the shim is different to the one above and always uses pkg_resources

Maybe this isn't A1 priority but I wondered about fixing it because it wasn't v obvious what was going wrong - I just had a dev environment which wouldn't work and failed with a "cannot find _kernel.." error*. Not sure what can be done to fix this though - perhaps there's a way of getting setuptools to create a custom .pyw file somehow. The standard .pyw file seems to be autogenerated when you use the gui_scripts entry point in the qt\applications\workbench\setup.py file.

DannyHindson, 2022-02-03:
I think this is an issue only for developers where the workbench.exe\workbench-script.pyw and egg-link's are relied upon to access the MantidWorkbench package in the source directories from within build\bin

DannyHindson, 2022-02-03:
The sitecustomize.py mechanism previously set up to handle setuptools>=49.0.0 seems to clash on Windows with a pre-existing sitecustomize.py that lives in C:\mantid\external\src\ThirdParty\lib\python3.8\Lib\site-packages. I don't think you can have two sitecustomize.py
I believe the sitecustomize.py was aiming to add the build\bin directory to the site directories which means the easy-install.pth file gets read (containing path to MantidWorkbench). For developers I think we can just remove the sitecustomize.py to fix this because the egg-link's do the work of pointing at the folder where the MantidWorkbench package lives.
I'm not sure if the sitecustomize.py has any purpose in a Mantid installation

DannyHindson, 2022-02-04:
So in summary two changes:
1) change qt\applications\workbench\setup.py so it generates a shim file that correctly finds the MantidWorkbench package
2) stop generating the sitecustomize.py in build\bin

With these changes, the only bit that doesn't work is the MantidWorkbench.exe (which is Windows\OSX only). I've only tested in on Windows but it's not finding the workbench.app.main path in a developer set up. Maybe this should use the gui_scripts entry point as well?

martyngigg, 2022-02-07:
> So in summary two changes:
> 
>     1. change qt\applications\workbench\setup.py so it generates a shim (.pyw) file that correctly finds the MantidWorkbench package. This can be done by using the scripts keyword to put some hand-crafted code in the shim rather than the default
> 
>     2. stop generating the sitecustomize.py in build\bin
> 
> 
> With these changes, the only bit that doesn't work is the MantidWorkbench.exe (which is Windows\OSX only) in a developer set up. I've only tested in on Windows but it's not finding the `workbench.app.main` path in the source directories. I guess this must have relied on the site.py mechanism with the older setup tools version. Maybe this should use the gui_scripts entry point as well?

Ah I see now. Yes, I believe that Python just does an `import sitecustomize` so whatever it finds first will take over. Would another option be to keep the one in `bin/<Config>` and remove the one from Third party? We're generating `bin` anyway so can add the contents that is already in the other one to there? This will no longer be necessary once we do the Conda switch.

martyngigg, 2022-02-08:
Yeah happy to review any pull request. It sounds like something that is heading in a promising direction.

DannyHindson, 2022-02-08:
I've just noticed there's been some quite recent discussion about this problem:

https://stackoverflow.com/questions/70809447/oddities-when-pip-installing-a-local-package-with-prefix-and-e-potential-bug

https://github.com/python/importlib_metadata/issues/364

This discussion suggests the .egg-link files are being deprecated so maybe my solution isn't going to be very long lasting even if it works for now. These links instead suggest relying on the sitecustomize.py as you suggested. I'll see if I can work out a way of getting the `bin/<Config>/sitecustomize.py` found reliably.

DannyHindson, 2022-08-18:
I think this is still an issue. For example this change was put through in earlier in this (6.5) release cycle to pin setuptools to 47.0.0:
https://github.com/mantidproject/mantid/pull/34134

DannyHindson, 2023-03-14:
Think this is still relevant

thomashampson, 2023-06-09:
This is a blocker for moving to python 3.10

closed 2023-07-24 by SilkeSchomann