mantidproject/mantid #33129: Saved focussed data doesn't appear to be correctly calibrated in EngDiff UI

RichardWaiteSTFC, 2021-12-14:
**Describe the bug**
Saved focussed data doesn't appear to be correctly calibrated in EngDiff UI: the North and South bank spectra should line up in TOF and d-spacing post calibration.

After a run is focussed the resulting workspace looks like this (zoomed in on one peak)
Saved focussed data doesn't appear to be correctly calibrated in EngDiff UI
[image]

But loading the data into the fit browser produces this
[image]

**To Reproduce**
Follow steps in manual testing docs to calibrate and focus data in tyhe EngDiff Ui
https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

**Expected behavior**
the North and South bank spectra should line up in TOF and d-spacing.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

DannyHindson, 2021-12-15:
I just tried this on my Windows install using main and I see the same thing. I created a fresh calibration using 30738 and then focussed using sample 305761 and V 307521 and got this from the fit tab:

[image]

The bank 2 peak seemed to be offset to the left in the plots produced on the focus tab for me as well

RichardWaiteSTFC, 2021-12-22:
Closed via #33153

closed 2021-12-22 by RichardWaiteSTFC