mantidproject/mantid #33154: Ideas for making Mantid easy to Cite

robertapplin, 2021-12-17:
**Description:**
The ‘Problem of Recognition’, outlined in this link https://medium.com/openteams/the-problem-of-recognition-fb9facfb4177  , is a hot topic in the software developer community. At the moment, there is no agreed-upon solution to this problem anywhere in the community, but it is possible that small improvements can still be with this goal in mind.

@martyngigg @sf1919 and I were discussing ideas for how we can make improvements within Mantid related to this problem. One small step towards improving recognition of the work done by our software developers could be to make it as easy as possible for Mantid to be cited in a scientific paper, thereby increasing the likelihood that our work gets cited.

This issue is for coming up with ideas that will make Mantid easier to cite.

Pasarus, 2021-12-17:
Check out this, we already have a lot of leg work on this issue with regards to citation:
https://github.com/mantidproject/mantid/issues/26159

mducle, 2021-12-17:
I'm a bit confused - the linked article is about attribution of contributions to a software project rather than about scientific citation... 

Regarding the citation issue, I think Mantid is quite well setup compared with many project in that we have a project DOI and also DOIs for each versioned release. To make it easier to cite, there's the [Citation File Format](https://citation-file-format.github.io/) which is now supported by github - it comes up as a link below the readme and license in the "About" section which would generate a citation text for you (also in bibtex format), see e.g. [its repo](https://github.com/citation-file-format/citation-file-format).

Regarding fully attributing all the contributions (including non-code contributions) to Mantid, it's a bit trickier but at least Mantid is a relatively small project so it's a lot easier than say `scipy`. A few ideas which come to mind:

* Use the list of code contributions to the script repository - this would cover most instrument scientists or "power" users who contribute scripts but don't have the inclination to contribute a PR.
* List all the people who have responded to forum questions
* List all the people who have participated in project management and review boards (Mantid seems to be very good about putting most of these online already).

robertapplin, 2021-12-20:
Apologies @mducle , I think this issue could have been explained better. The discussion I had with Martyn and Sarah revolved
around the contributions made to scientific software and how they are recognised, which is probably better described in this 
link and its adjoining article https://bssw.io/blog_posts/the-contributions-of-scientific-software-to-scientific-discovery. From our discussion, we identified two things that could potentially provide better recognition for the work done by our
software contributors:

1) Try to increase citations of Mantid for publications
2) Try to increase awareness of developer contributions

I think you mentioned these both in your comment. This issue is meant to focus on number 1 only (which isn't clear from the initial description of this issue).

Regarding the Citation File Format you linked, I think it looks promising and is certainly something that could help with
point 1 - so thanks for mentioning that! I will look into how this can be deployed within the Mantid repo.

For Mantid Users who don't look at github, it would be useful to have a similar mechanism for copying a Mantid citation
within workbench itself. This issue will therefore focus on two things:

A. Deploying the Citation File Format in the main Mantid Repo
B. Coming up with designs for a mechanism to easily get a Mantid Citation from within Workbench

If you have any other ideas or comments then please let me know

mducle, 2021-12-20:
Ah right, thanks for the clarification. So regarding your "B" point, these are some stuff I've seen other codes do:

* Include a "Please Cite" / DOI on initialisation (quite common) and Mantid does this if you do `import mantid.simpleapi` or otherwise initialise the Framework.
* Include a "Please Cite" / DOI at the end of each command - this is more common in packages that are made of different commands that you just run in a shell and can get quite annoying if we do it e.g. whenever someone runs an algorithm.
* Include a "Please Cite" blurb on the web documentation (very common) and Mantid also already does this.

I guess what I've not seen is something which generates the `bibtex` entry for you like the github CFF integration, but if you're writing a paper this is not really a blocker on you citing the program or not... (it's really easy to just create a bibtex entry or if you use a bibliographic manager like Zotero they will do it for you).

Now, as for getting people to cite the program... well, I think there's only so much we can do, but from my experience people are generally happy to cite software if they feel they get enough "value-added" out of it. For example, pretty much every paper I've seen from my users have cited Mantid in the "experiment methods" sections when they talk about how the data was collected. This also goes with [Horace](https://horace.isis.rl.ac.uk/) and [SpinW](https://spinw.org). A counter example is the [Vesta](https://jp-minerals.org/vesta/en/download.html) drawing program which draws crystal structures - it's basically the de facto standard and everyone uses it but only maybe 10-20% cite it, even though it states in the license that you must cite it if you generate images in a paper using it. I guess this comes down to people thinking that just making a picture doesn't constitute enough "value-added" to cite... for me personally I tend to _not_ to use Vesta _because_ of the license which I think it is a bit coercive... 

One final thought on open data / reproducibility - at the moment most people I know just cite the main Mantid DOI, but for reproducibility purposes we should try to encourage people to cite the actual version of Mantid they used (especially since we already have DOIs for each version).

robertapplin, 2021-12-21:
Thanks again @mducle for some more good advice. I can see that https://pypi.org/project/cffconvert/ provides some useful conversions that can be used for point B, so we can just take all the citation information we need from the one CFF file - nice!

I agree, the software version should definitely be included in the citations we provide

robertapplin, 2021-12-23:
For point "B" above, here is an example mockup design for easily copying a Mantid citation. Ignore the format of the displayed citation as it is just an example
[image]

In the Settings, we could have an option to save a custom citation format that they can reuse:
[image]

AnthonyLim23, 2022-01-05:
The easy citation work that @Pasarus linked should fix the problem of collaborators getting recognition. The documentation for each algorithm has a citations section (currently it is rarely used). When the user asks for the "citations" of a workspace it will look at the algorithm history and pull out the corresponding citations plus the Mantid citation. So all collaborators will need to do is make sure that they provide a citation in the docs.

AnthonyLim23, 2022-07-07:
I would still like to see this done

closed 2023-02-04 by stale[bot]