mantidproject/mantid #33164: Instrument view pick tab erroneously plots sum of all spectra by default

RichardWaiteSTFC, 2021-12-22:
**Describe the bug**
Found during review of PR #33161 
Instrument view appears to plot all spectra and subsequently adds all peak labels when you draw a ROI then delete it (or when you click on the magnifying glass on the pick tab - probably other things too would cause this)

**To Reproduce**
To-Do

**Expected behavior**
It would not plot anything until a ROI drawn or a spectrum selected/cursor over a spectrum 

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

DannyHindson, 2022-01-06:
It seems like this was deliberately added as a feature in this PR:

https://github.com/mantidproject/mantid/pull/32014

...and specifically in this commit:

https://github.com/mantidproject/mantid/commit/5082d641b48b898452b1e09e9c676ee31ac0aa54

@RichardWaiteSTFC - what is it you don't like about this? I can double check with Mathieu\Gagik what their thinking was but would be useful to have some reasons to back up a suggestion to revert this

RichardWaiteSTFC, 2022-01-07:
> It seems like this was deliberately added as a feature in this PR:
> 
> #32014
> 
> ...and specifically in this commit:
> 
> [5082d64](https://github.com/mantidproject/mantid/commit/5082d641b48b898452b1e09e9c676ee31ac0aa54)
> 
> @RichardWaiteSTFC - what is it you don't like about this? I can double check with Mathieu\Gagik what their thinking was but would be useful to have some reasons to back up a suggestion to revert this

@DannyHindson - I had a vague recollection it was an ILL change, I think Gagik mentioned it in the dev meeting. I think the change is not particularly useful for our instruments (possibly it is for instruments with detector banks all at similar tth such as ENGIN-X, but they don't use the instrument view anyway). The only reason I want to change it is because for the case of WISH the summing of the entire instrument appears to be causing a slow-down, even on IDAaaS apparently.

DannyHindson, 2022-01-10:
@MathieuTillet @gvardany - please can you have a look at this bug that's been raised during our manual testing. It relates to a feature that was intentionally added during PR #32014. The fact that the instrument viewer now performs the sum across all detectors when no shape has been drawn has a couple of downsides for us:

1) this causes a delay of 1-2 seconds when switching between the buttons on the Instrument Viewer Pick tab or when you delete a shape
2) when there is a peaks workspace overlaid on the main workspace, the hkl values for all peaks are shown which is a bit messy for an initial view because they overlap (see example screenshot higher up this issue). If a user has drawn a shape that intentionally captures a lot of peaks then the view may still show overlapping labels like this but the user may be expecting this a bit more

We're wondering whether to revert this part of #32014. Do you have a strong opinion on this? What was the requirement you had that triggered this change?

DannyHindson, 2022-01-14:
@gvardany - just wanted to follow up on this. Did you see my question about whether we can revert a small piece of #32014 (the auto summation of all the detectors)? I wasn't sure whether one of your users had specifically requested this

gvardany, 2022-01-17:
Sorry for catching up so late, the notification got lost in the plethora of other emails. I'll examine the story and let you know. I can tell now, that summing (but not overplotting) of all the spectra even when there is no ROI drawn is a requirement. However, this is for MatrixWorkspaces, does your use case concern PeakWorkspaces? (I must say I am not terribly familiar with those).

martyngigg, 2022-01-18:
Thanks for replying with the information. Danny's away so I'll follow up.

Our case is also for `MatrixWorkspace`, it's just that the peaks overlay is what caused us to notice it. The default since the creation of the instrument view has been to plot pixels/tube/bank, depending on the 2 lower-left buttons selection in the pick tab, in the miniplot as the cursor moves. If we move the plot only everything then we can't inspect individual tubes?

As the change is also not a visible one I think changing the default is an issue. Could we add an extra tool button next to `select whole tube` button that would be `select whole instrument` instead? We could also always had a configuration option that would allow that to be the default tool?

gvardany, 2022-01-28:
I'm fine with having another button for selecting the whole instrument, as long as one doesn't need to draw any shape, but just click on that button. @MathieuTillet any objections from you? I think you know that use case the best.

martyngigg, 2022-01-28:
> I'm fine with having another button for selecting the whole instrument, as long as one doesn't need to draw any shape, but just click on that button. @MathieuTillet any objections from you? I think you know that use case the best.

Thanks. That works for me. If it's the whole instrument button then having to draw something doesn't make too much sense anyway.

MathieuTillet, 2022-01-28:
That looks like a fair compromise to me.

PR #33421 (2022-02-02): Add new button to Instrument View Pick tab
**Description of work.**

Add new button that automatically selects the whole instrument for summation without having to draw a shape. This resolves a clash between the new requirements implemented on #32014 and the desire from some ISIS users not to have a short delay when switching buttons on the Pick tab.

In order to avoid creating a new input controller I have reused the move controller (used for the Navigate button) for when the new "Whole instrument" button is pressed. Perhaps this is helpful anyway given that the earlier implementation had the [... cut]