mantidproject/mantid #33185: Manual Testing ISIS Indirect interfaces

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow the directions at: https://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

DannyHindson, 2022-01-05:
Notes from our ensemble testing:
```
Testing Docs:
No energy transfer instructions

(ISIS Calibration tab): 
Mention reflection is in the top right
Ask a user to press run after changing reflection

(ISIS Energy Transfer tab):
Make it clear we're testing the energy transfer tab
Step 11, change from "cancel" to, "pick a random spectra and plot this" (include a picture?)

-----------

Bugs - ISIS Calibration Tab:
Normalisation with scale 0.98 are 1.0 are very different
Load Calibration, first graph is not changing at all
Loading another file causes the fit properties to vanish so create RES does not work  (59057-59059 and check Sum Files, Reflection 004)

Bugs - ISIS Energy Transfer:

-------------------------------------

Data Analysis Elwin

Add note to switch to Data Analysis UI
Input File area was massive initially
We updated the search dirs after having populated input file and when we hit run it didn't seem to be revalidating the input file - still had red star

MSD

-----------------------------

Not clear that you load file vai the "Add Workspace" button
Unclear that you need to check the "All Spectra" checkbox
Make it clear that it's set the Fit Type to Peters and Yi
We used Damped GaussNewton and it produced a huge chi squared that messed up the width of the fit property browser. Chisq was too wide

Conv Fit
--------
Set Fit spectra to 0 - 5. Clarify how to do this
"In the Fit tab, change Fit spectra to String and enter 3" - this instruction isn't clear in the new UI. Seems to be the case that you now click "single spectrum"
Perhaps change button text to "Fit Selected Spectrum" or similar
** We had a crash while deleting some rows from the spectrum list at the top - this was the second removal, specifically removing 0 as part of the second removal action**

"should give 4 separate plots" - we got 5 plots

I(Q,t)
------
Could we go in order of the tabs? This is before ConvFit for example
"A new workspace with the suffix _iqt should appear in the main GUI, it should be a 87 x 6 table. NB keep this workspace for the next test" Ours is 86 x 51
The Plot Spectra button takes a list of workspace indices

I(Q,t ) Fit
-----------
Clarify instruction - Need to hit Close on the Add Workspace window
Clarify instruction - Exponentials rt Exponential
Same crash as in Conv Fit tab when doing double remove

Nice to add some tests for the other tab in Data Analysis UI: F(Q) Fit
```

Harrietbrown, 2022-01-10:
Crash on data removal has been fixed by #33220

DanielMurphy22, 2022-01-11:
@Harrietbrown are there any issues that should be raised for the bugs / niggles remaining? Otherwise are we able to close this issue?

Harrietbrown, 2022-01-11:
@DanielMurphy22 I will be make a few issues then it can be closed.

#33266

Harrietbrown, 2022-01-11:
#33267

Harrietbrown, 2022-01-11:
The other findings are so minor I don;t think they require issues for.

closed 2022-01-11 by Harrietbrown