mantidproject/mantid #33186: Manual Testing Documentation

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

In the build directory run `cmake docs-html` 
Check in the `docs/html` directoy:

* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

You may wish to use this script: [OpenMostDocumentationForTesting.py](https://github.com/mantidproject/mantid/blob/master/tools/scripts/OpenMostDocumentationForTesting.py) to open all the online docs pages!

sf1919, 2022-01-06:
Are the specific notes correct? The first bullet point says to use installer versions not a local build so why are we running `cmake docs-html`?

sf1919, 2022-01-06:
The  OpenMostDocumentationForTesting.py  file needs updating as it still points to Mantid Plot in several places.

sf1919, 2022-01-06:
Issues
Concepts - In the Concepts directory the first option is '02_scripting_workspaces` which looks incorrect

Print to PDF crash -  When we tried to print to Cute PDF from the Help it crashed.

sf1919, 2022-01-06:
Otherwise everything works fine and the images, scripts etc that we looked at all displayed well.

closed 2022-01-06 by sf1919