mantidproject/mantid #33187: Manual Testing Workbench Group 1- 4

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

DavidFair, 2022-01-06:
Mac OSX Bugs:
- `Default Facility` and `Default Instrument` does not scale (especially noticeable with empty instrument)
-

DavidFair, 2022-01-06:
All:
- Sliceviewer, enabling line plots and dragging a ROI box over it will try to use the spectrum / TOF values on the line plot causing weird ROI box behaviour @RichardWaiteSTFC

thomashampson, 2022-01-06:
Ubuntu and IDAaaS bugs:
- Changing min or max y-axis limits (or the tick format) on a 3D plot (surface or contour) will work but display an error message in the dialog:
[image]

DanielMurphy22, 2022-01-06:
> Ubuntu and IDAaaS bugs:
> 
> * Changing min or max y-axis limits (or the tick format) on a 3D plot (surface or contour) will work but display an error message in the dialog:
>   [image]
> 
> This appears to be linked to MPL 2.x versions

https://github.com/mantidproject/mantid/issues/30014

thomashampson, 2022-01-06:
Plot to script button is missing on tiled plots if it's not a square number of plots:
[image]

DannyHindson, 2022-01-07:
A few script repository issues. Unclear if these are worthy of issues:

1) can't delete a file that's been uploaded by me (+ typo in the warning message):
[image]

2) error on trying to upload a folder eg created a new folder for me in the users directory. Message says recursible upload not possible. Perhaps this isn't meant to be possible and you just upload the contained files. Uploading the contained file did create the containing folder for me but did give an error...

3) Following upload of the new file in my new user directory, I had a .json error relating to the .repository.json file reported in the messages log (this was on Windows) - but went away went restarting workbench. Mial was testing on IDAaaS and didn't get this. Didn't repeat because didn't want to create too many new folders in script repository that I couldn't subsequently delete

DanielMurphy22, 2022-01-07:
> 3. Following upload of the new file in my new user directory, I had a .json error relating to the .repository.json file ...

I wonder if the error will be removed after our fix to JsonHelpers::jsonToString https://github.com/mantidproject/mantid/pull/33243

DannyHindson, 2022-01-10:
We found this problem with the scripting of workspace history. The workspace was generated using the ISIS Powder diffraction scripts eg script the history on workspace 98533-ResultD generated by running this:
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

from isis_powder import polaris, SampleDetails

config_file_path = r"C:\Total Scattering\polaris_config_example.yaml"
polaris = polaris.Polaris(config_file=config_file_path, user_name="test", mode="PDF")

sample_details = SampleDetails(height=4.0, radius=0.2985, center=[0, 0, 0], shape='cylinder')
sample_details.set_material(chemical_formula='Si')
polaris.set_sample_details(sample=sample_details)

polaris.create_vanadium(first_cycle_run_no="98532", multiple_scattering=False)
polaris.focus(run_number="98533", input_mode='Summed')
```
There are some calls to algorithms during the V normalisation (RebinToWorkspace, Divide) that set `StoreInADS` to false and these don't seem to be scripted properly

DannyHindson, 2022-01-10:
Instrument Viewer issues:

- crash on Draw tab: https://github.com/mantidproject/mantid/issues/33232
- screen goes black when zooming in to within a detector bank on Pick tabs. This was present in 6.2 though so probably won't create an issue
- Can't use the Navigation tool on the Pick tab if you're using the "Full 3D" view. The mouse clicks that are made in any attempt to draw a zoom rectangle get interpreted as a request to rotate the shape. This was also present in 6.2 so won't create an issue

DanielMurphy22, 2022-01-11:
> Mac OSX Bugs:
> 
> * `Default Facility` and `Default Instrument` does not scale (especially noticeable with empty instrument)
Is this on the About MantidWorkbench / First time setup menu? Did it look like this? 
[image]

> * Plotting: Script to clipboard does not paste anything into clipboard
I cannot reproduce on BigSur 04012022nightly, it works fine for me! Any tips to reproduce?

DannyHindson, 2022-01-11:
> > Mac OSX Bugs:
> > 
> > * `Default Facility` and `Default Instrument` does not scale (especially noticeable with empty instrument)
> >   Is this on the About MantidWorkbench / First time setup menu? Did it look like this?
> 
> [image]
> 
Yes it does relate to that screen and my memory is that the instrument drop down did look the same as you've got there. The instrument dropdown was too narrow for the user to see the instrument names in the list properly. Don't remember the second scripting issue so will need to leave that to @DavidFair to comment on

DanielMurphy22, 2022-01-11:
All issues raised

closed 2022-01-11 by DanielMurphy22