mantidproject/mantid #33188: Manual Testing Muon

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

AnthonyLim23, 2022-01-05:
Found a bug: load some data, the plot will round all decimals to be 0

AnthonyLim23, 2022-01-06:
the testing pages we should look at are (i.e. not completely out of date):

FDA: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_4_FDAG.html 
MA (PSI): https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_test_group_5_MA2.html
MA (ISIS): https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html
ALC:  https://github.com/mantidproject/mantid/pull/29942 and https://github.com/mantidproject/mantid/pull/26256 and https://github.com/mantidproject/mantid/pull/25089

AnthonyLim23, 2022-01-06:
FDA: explicit say that its the x min and max to set

AnthonyLim23, 2022-01-06:
PSI say to look in BG col

AnthonyLim23, 2022-01-06:
be consistent in https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

AnthonyLim23, 2022-01-06:
TF asym crash - to do with errors

closed 2022-01-10 by martyngigg