mantidproject/mantid #33191: Manual Testing Engineering Diffraction

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

sf1919, 2022-01-07:
We have completed all the testing bar one small thing (issues to be listed separately). We still need to test whether using an RB number string with a space in it causes a crash on non-Windows systems. This is because a folder is created using the RB number string as the name. It works fine on Windows but need to check other systems.

sf1919, 2022-01-07:
Issues

Testing documentation
- update image for the calibration tab
- update umage for focus tab
- under test 5 need to add instructions on how to load multiple runs

Table in Fitting
- We encountered an error when we removed a row from the table (which worked properly) and then clicked the remove all button. There was a crash with the following stack trace
 ```
Error occurred in handler:
Traceback (most recent call last):
File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\fitting_ads_observer.py", line 20, in wrapper
 func(*args, **kwargs)
 File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\fitting_ads_observer.py", line 75, in replaceHandle
 self.replace_callback(name, workspace)
 File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_widget.py", line 40, in replace_workspace
 self.presenter.replace_workspace(name, workspace)
 File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_presenter.py", line 105, in replace_workspace
 self._repopulate_table()
 File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_presenter.py", line 139, in _repopulate_table
 self._remove_all_table_rows()
 File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_presenter.py", line 290, in _remove_all_table_rows
 self.view.remove_all()
 File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_view.py", line 189, in remove_all
 self.table_selection.setRowCount(0)
RuntimeError: wrapped C/C++ object of type QTableWidget has been deleted
```

Plots in Fitting
- add a legend to Background plot
- Input Background plot not auto updating when background parameter in UI table changed (definite bug)

Model Info table
- Model info table for a fit not updated for a fit of a textured spectrum

On top of this we ran into a strange crash when trying to close the About Mantid popup that seemed to be a problem with the config file. We've been unable to recreate it but here is the stack trace for ref
```
Traceback (most recent call last):
 File "C:\MantidNightlyInstall\bin\workbench\widgets\about\view.py", line 47, in closeEvent
 self.presenter.save_on_closing()
 File "C:\MantidNightlyInstall\bin\workbench\widgets\about\presenter.py", line 198, in save_on_closing
 self.parent.config_updated()
 File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 701, in config_updated
 self.editor.load_settings_from_config(CONF)
 File "C:\MantidNightlyInstall\bin\workbench\plugins\editor.py", line 59, in load_settings_from_config
 self.editors.load_settings_from_config(config)
 File "C:\MantidNightlyInstall\bin\mantidqt\widgets\codeeditor\multifileinterpreter.py", line 93, in load_settings_from_config
 self.confirm_on_save = config.get('project', 'prompt_save_editor_modified')
 File "C:\MantidNightlyInstall\bin\workbench\config\user.py", line 94, in get
 raise_keyerror(full_option)
 File "C:\MantidNightlyInstall\bin\workbench\config\user.py", line 87, in raise_keyerror
 raise KeyError('Unknown config item requested: "{}"'.format(option))
KeyError: 'Unknown config item requested: "project"'
```

RichardWaiteSTFC, 2022-01-10:
Closing this issue as testing done - @DannyHindson confirmed that RB folder is allowed to have a space in it on Linux.

closed 2022-01-10 by martyngigg