mantidproject/mantid #33194: Manual Testing ISIS Reflectometry

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

For testing procedures see http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

AnthonyLim23, 2022-01-05:
When autoprocess is complete, it doesn't seem to send a completed signal correctly as the GUI gets stuck in the processing state

AnthonyLim23, 2022-01-05:
Transfer table does not work when there is no theta specified in the title

AnthonyLim23, 2022-01-05:
cannot paste into unselected position in the processing table - better error message (e.g. "please select a group to paste the data into")?

AnthonyLim23, 2022-01-05:
Fill down only works for run number

AnthonyLim23, 2022-01-05:
when closing GUI it asks if the user wants to close it, but does not have "save, cancel and ok" options

AnthonyLim23, 2022-01-05:
warnings are not on by default

gemmaguest, 2022-01-05:
> cannot paste into unselected position in the processing table - better error message (e.g. "please select a group to paste the data into")?

I've updated this issue with a note about improving the error message or making this more intuitive #23067

gemmaguest, 2022-01-05:
> when closing GUI it asks if the user wants to close it, the docs need updating to reflect this

PR #33209 created to improve the docs

gemmaguest, 2022-01-05:
> warnings are not on by default

I think they should be in a fresh install (I had probably previously disabled them in my idaaas session) but I will check.

closed 2022-01-10 by martyngigg