mantidproject/mantid #33195: Manual Testing ISIS SANS (new GUI)

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://developer.mantidproject.org/Testing/SANSGUI/SANSGUITests.html

DavidFair, 2022-01-06:
Notes from testing:

- [ ] Load user file is case sensitive .TOML
- [ ] Pasting multiple items reverses the order relative to clicked order
- [ ] **Uncaught exception if column order is not expected**
- [ ] CSV parsing error: `row n seems to be missing` shows (n-1)
- [ ] Load User File, loading *.txt resets file extension filter to a wildcard (RHEL only?)
- [ ] Plot Results with Merged results in 8 plots, instead of 6 as HAB is duplicated

DanielMurphy22, 2022-01-11:
@DavidFair do we need to raise any issues for the bugs found or do some issues already exist?

DavidFair, 2022-01-12:
I was opening issues as / when tackled as all of them are niggles.

Lets close this issue as the release could ship with all of the above perfectly fine (they're weird edge cases except for #33278 ) and the testing is done

closed 2022-01-12 by DavidFair