mantidproject/mantid #33197: Manual Testing Mantid Basics Course

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Build a docs-html target of Mantid
* Build docs-qtassistant and docs-qthelp so that autogenerated plots are available
* Check against the online documentation (http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/index.html#mantid-basic-course)
* Open up the basic course (docs/html/tutorials/mantid_basic_course/index.html)
* Check that the pages in there make sense

AnthonyLim23, 2022-01-07:
Just check the doc:

- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/getting_started/getting_started.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/getting_started/main_window.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/index.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/06_formatting_plots.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/index.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/01_introduction.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/01_introduction.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#save-settings
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/02_scripts.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/03_user_support.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/04_the_end.html

Check the doc and do the instructions

- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/01_loading_data.html#loading-data
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/02_the_matrix_workspace.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/05_plotting_advanced.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/07_exercises.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/03_workspaces.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/04_algorithm_histories.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/05_interfaces.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/06_exercises_a.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/02_displaying_and_navigating.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/04_masking_and_grouping.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/05_instrument_tree.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/06_sample_logs.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/07_invalid_sample_logs.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/08_exercises_c.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/02_complex_case.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/04_intelligent_fitting.html
- https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/05_fitting_exercises.html
-

AnthonyLim23, 2022-01-07:
I count 12 where we just need to read the docs and 22 that need reading and testing

AnthonyLim23, 2022-01-10:
https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html

- the toolbar needs updating

ConorMFinn, 2022-01-10:
Loading a range of runs into a group and loading and summing a range of runs both have the same default output workspace name (`MultiFiles`). This means that anything loaded this way is overwritten without warnings when doing any load of a range with the default options.

ConorMFinn, 2022-01-10:
The Displaying 2D Data page makes reference to a "boring purple display" that is no longer the case after some changes to the MARI data file handling.

AnthonyLim23, 2022-01-10:
I can crash mantid by:
1. Plot some data and go to do a fit
2. Add a user function
3. Click fit (Without giving a function)
4. Mantid will complain in the logger
5. Add a function to the user function by using the ...
6. Press fit - crash

MialLewis, 2022-01-10:
Script to clipboard works on IDAAS inconsistently (ctrl + v does not work, right click -> paste generally does.

Copy and paste shapes on instrument viewer behaves inconsistently 
- Shapes that are pasted in are highlighted as if they're selected, but are actually not. The original shape that is copied remains selected. Preferred behaviour would be to deselect the originally copied shapes, and highlight the newly inserted shape.
- The above does not apply to the rectangular ring shape, which is just not highlighted at all.
- Shapes that are pasted in are inserted at a location a set x and y position from the copied shape. It is possible for the shape to be inserted off-screen. Ideally, the paste offset would be smaller, and would not be pasted off-screen.

AnthonyLim23, 2022-01-10:
https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/getting_started/getting_started.html

- better wording:
"On the downloads page you will be able to get the current stable release of Mantid, which is recommend. There is also an option to download the latest nightly build, this is a more experimental version of the software. Whilst the nightly build may have more features, it is less tested and comes is likely to be less stable than the release."
- the error reporter image needs updating

DanielMurphy22, 2022-01-10:
Connecting Data to Instruments section

- index.html has no foreword, maybe because of the introduction page. Maybe could link to some facility pages?
- 01 introduction page, the 3 elements and tabs should just be labelled on the image!!!
- 02 Say the **FULL** 3D view; zoom by "rotating the wheel" -> scrolling
- 03 Explain Phi; the zoom tool doesn't work in full 3D; Edit shape tool autoselects after drawing a shape; reword the "rubber band" comment.
- 04 is quite boring! Explain the integration bar better (bottom of page)
- 05 Instrument tree - "drill down into detector structure through detector banks to pixels". 
- Exercises: explain 5e4; tooltips for the tool icons on pick + draw tab have keyboard shortcuts that say Ctrl, but use Cmd on Mac!

Other notes:
- Anthony comment: "The parameter value obeys the constraint within the error"
- Mial comment on Alg History: When using cloud/virtual machines the clipboard may not work in both directions. Possibly use right-click paste is better than Ctrl+V
- Conor Comments: Double clicking the colorbar on a colorfill gives the figure options, but not changed to the images tab!
- Daniel Comment: breadcrumb!!!!!

DanielMurphy22, 2022-01-10:
double it on page: https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#save-settings. 
"Within the Mantid": https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/02_scripts.html

closed 2022-01-10 by DanielMurphy22