mantidproject/mantid #33198: Manual Testing M-slice

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://mantidproject.github.io/mslice/ and follow Quick Start, this should probably be tested in iDAaaS

ajjackson, 2022-01-06:
Following the testing guide at https://developer.mantidproject.org/Testing/MSlice/MSliceTestGuide.html

(Link in the Issue description goes to main MSlice docs)

ajjackson, 2022-01-06:
Comments on actual test guide
- It would benefit from some screenshots of expected results.

DanielMurphy22, 2022-01-06:
> (The link in the Issue description goes to main MSlice docs.)

I will update this is the spreadsheet used to generate the testing issues!

closed 2022-01-06 by martyngigg