mantidproject/mantid #33199: Manual Testing Project Recovery

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

AnthonyLim23, 2022-01-06:
we should add to the docs to check that this works 
```
Load(Filename='/archive/ndxmusr/Instrument/data/cycle_16_5/MUSR00062260.nxs', OutputWorkspace='MUSR00062260.nxs', TimeZeroList='0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474,0.474', DeadTimeTable='MUSR00062260.nxs_deadtime_table', DetectorGroupingTable='__notUsed')
```

DanielMurphy22, 2022-01-06:
- The setting should for ProjRec have a tooltip (not there on IDAaaS, at least)
- Should ProjRec report to the logs the a restart is needed for the setting to be obeyed
- The report usage data checkbox on the About Workbench First time startup menu should maybe be in settings and be clear that it also affects Error Reporting!
- Low priority: Could settings and paths to open scripts be saved by project recovery? Currently settings rely on Workbench successfully closing to be enabled.

**Testing Docs**
- Rewrite section 3 to use an ISIS file (rather than TOPAZ) and put the new file in the ISIS Sample Data
- Add comment to also open Engineering Diffraction in section 4

DanielMurphy22, 2022-01-11:
> we should add to the docs to check that this works
>
> ```
> Load(Filename='/archive/ndxmusr/Instrument/data/cycle_16_5/MUSR00062260.nxs', OutputWorkspace='MUSR00062260.nxs', [...]

@AnthonyLim23  It totally escapes me what test instruction should be added here for this! Can you add a bullet point to the bottom of this issue please? https://github.com/mantidproject/mantid/issues/33271

DanielMurphy22, 2022-01-11:
All worthy issues raised

closed 2022-01-11 by DanielMurphy22