mantidproject/mantid #33201: Manual Testing Sliceviewer

DanielMurphy22, 2022-01-05:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

martyngigg, 2022-01-05:
# SliceViewer

## Issues

- Testing Docs:
  - Split single page up by workspace type
  - Each page of instructions should be easy to follow from top to bottom without jumping around
  - Use either links or include statements to share common sections
  - Add item to 
  - Section 8:
    - ROI: Check when ROI enabled that line plots button is enabled
  - Do delete workspace as last check
 
 - Autoscaling:
    - MatrixWorkspace: Inconsistent behaviour of autoscaling when zooming and clicking Home
      - Different numbers after unchecking/rechecking autoscaling
    - MDWorkspace (2D): Scale not updating as you zoom  
    - **Needs fixing for 6.3**
 
 - Enable line plots:
   - Resize to extreme aspect-ratio, labels from main X/Y plots are still visible
   - Minor, not for 6.3
 
 - Cursor info widget not fixed-width formatting so widget changes size
   - Minor, not for 6.3 

- Deleting peaks workspace does not close the peaks viewer
  - No issues exists

- Deleting workspace does not close the window

- 2D histo: Transposing gives an error, but tranpose appears to have worked correctly:
```
mdHisto_3D = BinMD(InputWorkspace=md_4D, 
    AlignedDim0='H,-1,1,100', AlignedDim1='K,-1,1,100', AlignedDim2='L,-1,1,100') # 3D cut
mdHisto_2D = BinMD(InputWorkspace=md_4D, 
    AlignedDim0='H,-1,1,100', AlignedDim1='K,-1,1,100') # 2D cut
```
```
Traceback (most recent call last):
 File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\presenter.py", line 207, in dimensions_changed
 if sliceinfo.slicepoint[data_view.dimensions.get_previous_states().index(None)] is None:
ValueError: None is not in list
```

RichardWaiteSTFC, 2022-01-06:
Update docs to create this 4D MD workspace that contains some data with features that make transposing, slicing results more obviously correct...
```
md_4D_1 = CreateMDWorkspace(Dimensions=4, Extents=[-1,1,-2,2,-3,3,-4,4], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D_1, UniformParams='5e6') # 4D data
md_4D_2 = CreateMDWorkspace(Dimensions=4, Extents=[-0.5,0.5,-1,2,-3,2,-4,4], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D_2, UniformParams='5e6') # 4D data
md_4D = md_4D_1 + md_4D_2

expt_info = CreateSampleWorkspace()
md_4D.addExperimentInfo(expt_info)
```

closed 2022-01-06 by martyngigg

RichardWaiteSTFC, 2022-01-07:
Issue with wobbling cursor info table is already an issue #32358