mantidproject/mantid #33206: macOS package: An additional include directory has appeared when expanding the dmg

martyngigg, 2022-01-05:
**Describe the bug**
An extra `include` directory now appears when you open the macOS `.dmg` to copy the bundle to `/Applications`

**To Reproduce**
1. Download the current macOS build from Jenkins clean jobs 
2. Open the .dmg and the screenshot below is presented.

**Expected behavior**
There is just the MantidWorkbench application and the link out to the Applications directory.

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: macOS
 - OS Version: All
 - Mantid Version: Current nightly

**Additional context**

DavidFair, 2022-01-06:
Just an FYI `share` is also included, you need to drag a box towards the right of the window to see it too

martyngigg, 2022-01-06:
Thanks. It appeared more obviously on the command line.
It was difficult to convince cmake not to install them so I've just gone with deleting them in the post-install step now. Long-term this method will disppear with conda

PR #33216 (2022-01-10): Manually remove extra include/share directories from macOS bundle
**Description of work.**

Including eigen using FetchContent registers its install rules with
mantid. For macOS this resulted in the include/share directories
being copied to the root of the package and visible to users when
opening the dmg

See screenshot in #33206 for the error that this fixes.

**To test:**

* On macOS, enable cpack on your build and build the package:
```
build>cmake -DENABLE_CPACK=ON .
build>ninja package
build>open <name of .dmg file> 
```
See that there is only the `Applications` and `MantidWorkbench` lin [... cut]
closed 2022-01-10 by martyngigg