mantidproject/mantid #33214: Sliceviewer: Colorbar autoscaling not behaving as expected

martyngigg, 2022-01-06:
**Describe the bug**

During manual testing for 6.3 (#33201) it was disovered that colorbar autoscaling does not work as expected and provides an inconsistent experience for the user.

**This is a regression and must be fixed for 6.3**

**To Reproduce**

Use the data and step 1 from the [manual testing instructions](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#viewing-data). From manual testing it was observed:
 - MatrixWorkspace: Zoom in with autoscale checked, hit Home and the autoscaled value does not match the value you get if you then uncheck/recheck the box.
 - MDWorkspace (2D/3D/Histo): Scale not updating as you zoom

**Expected behavior**
* While enabled the colobar scale should update to the min/max of the viewed slice.

**Platform/Version (please complete the following information):**
 - OS: Tested on Windows
 - OS Version: 10
 - Mantid Version mantidnightly-6.2.20220104.1619

thomashampson, 2022-01-06:
Related to https://github.com/mantidproject/mantid/issues/32359

RichardWaiteSTFC, 2022-01-07:
Specific instructions to reproduce
**To Reproduce**

To test MAtrixWorkspace
1. Load `CNCS_7860_event.nxs` and open in sliceviewer
2. Make a note of the maximum color limit
2. Check autoscsale is on/checked
3. Zoom in - note colorscale will change
4. Click Home - colorscale will change but will not go back to original limit
5. Click Home - will change again, this time back to original limit

To test MDHisto zooming
(1) make the required workspaces
```
md_4D_1 = CreateMDWorkspace(Dimensions=4, Extents=[-1,1,-2,2,-3,3,-4,4], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D_1, UniformParams='5e6') # 4D data
md_4D_2 = CreateMDWorkspace(Dimensions=4, Extents=[-0.5,0.5,-1,2,-3,2,-4,4], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D_2, UniformParams='5e6') # 4D data
md_4D = md_4D_1 + md_4D_2

mdHisto_3D = BinMD(InputWorkspace=md_4D, AlignedDim0='H,-1,1,100', AlignedDim1='K,-1,1,100', AlignedDim2='L,-1,1,100') # 3D cut
mdHisto_2D = BinMD(InputWorkspace=md_4D, AlignedDim0='H,-1,1,100', AlignedDim1='K,-1,1,100') # 2D cut
```
(2) Open `mdHisto_3D` or `mdHisto_2D` in sliceviewer
(3) Zoom in - colorscale won't change when zooming (it should!)

gemmaguest, 2022-01-10:
This is reproducible in v6.0.0 so looks like it has been around quite a while. @martyngigg it's probably not a high priority issue because it hasn't been noticed for at least 3 releases! I'll continue to take a look to see if there's an easy fix.

gemmaguest, 2022-01-12:
**For Matrix workspaces:**

The problem seems to be that the colorbar updates the scale from the image data [here](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/colorbar/colorbar.py#L298) (the mappable here points to the image in the [view](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/sliceviewer/view.py#L313).

This gets called from [update_clim](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/sliceviewer/view.py#L351). However the image data does not get updated until the canvas is redrawn [here](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/sliceviewer/view.py#L357) (specifically at [this line](https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/mantid/plots/resampling_image/samplingimage.py#L127) in the `SamplingImage`, which is the type of image created by `imshow` for matrix workspaces).

If I reorder the `canvas.draw` line to be before `update_clim` then it seems to work ok but I don't think this is a good way to solve this. It also does not resolve the problem for MD workspaces.

This is also the cause of the problem in #32359. I get the same behaviour for event workspaces.

**For MD workspaces:**

The root of the problem is the same, in that the colorbar is trying to update the scale from the image data but the image data has not been updated. However, the image data never gets updated because MD workspaces use a `ModestImage` rather than a `SamplingImage`, which does not do resampling and does not seem to update the image data at all.

gemmaguest, 2022-01-13:
@thomashampson , @martyngigg Let me know if you think we should try to get this fixed for the release. My feeling at the moment is it's a missing feature and likely to be too much work to get in at this stage. (Although there could still be an "easy" fix I just haven't found yet.)

I'm happy to explain the problem over a chat if that's easier than deciphering my comment above.

martyngigg, 2022-01-21:
Fixed by #33336

PR #33336 (2022-01-21): Fix sliceviewer colorbar autoscale
This PR fixes two issues that are causing autoscale to fail for the sliceviewer colorbar:

- Move the `canvas.draw` to before calculating the colorbar limits, because the limits are based on the image data. For Workspace2D this is not recalculated before we redraw.
- For MD workspaces, add a new function to get the image data clipped to the axes extents (because the default `get_array` function always returns the full extent). Note that this works regardless of where the draw is done with the previous fix.

Fixes #33214

**To test:**

See the instructions in  [... cut]
closed 2022-01-21 by martyngigg