mantidproject/mantid #33215: Redirect log messages to Python logger

jl-wynen, 2022-01-06:
When using Mantid from Python, it would be useful if log messages were sent to Python's loggers instead of printing to terminal or file.

This would allow users to control if and how logs are shown and also combine them with logs from other software.
It should be fairly straight forward to configure Mantid's logger during import from Python to forward all messages to the Python logger.

I have already started looking into it because it would help us a lot in scipp. But I still wanted to inform you before submitting a PR out of the blue.
Do you have any comments and do you even think this is worth adding to Mantid?

jl-wynen, 2022-01-06:
Got a working implementation in my fork: https://github.com/jl-wynen/mantid/tree/python-logging

Before opening a PR, I was wondering what are your thoughts on setting up a logger?
It can be done such that if you import Mantid from Python, all channels are replaced by the new one which sends messages to Python's `logging`. But that might break existing setups.
Or we leave it up to the user and let them set a channel. This would be enough for us in Scipp. When calling Mantid functions, we can replace the logger ourselves. But it can lead to mixed log channels when users call both Scipp and Mantid functions themselves.

peterfpeterson, 2022-01-07:
After adding the `PythonStdOut` channel I had thought about moving completely over, but didn't have the time. This is a good addition, that needs a couple of small tweaks that would be better described in a PR review. I'm noticing that it is missing user documentation changes `concepts/PropertiesFile.rst` and am not sure that `log_to_python()` is a necessary function rather than instructing users on how to modify their properties file. On linux, there is a system level properties file that could set the logging channel for users on that machine as a whole.

Asking the question that was asked of `PythonStdOut` when it was introduced: Is this performant enough to be used as the default logging setup? I think it is a separate chunk of work to move, but I am curious if this could take over.

jl-wynen, 2022-01-07:
I am opening a PR and will answer to your comments about this particular change there.

> Is this performant enough to be used as the default logging setup?

Not sure. It has significant downsides: It always has to acquire the GIL when it receives a log message. That can add unwanted contention to multi threaded code. I don't know how much Mantid logs in parallel code. (Does Poco have a better solution?)

But more importantly, it requires that the interpreter is running. If you look at the destructor of `PythonLoggingChannel`, you can see that I had to use some dirty hacks to make it work.
Python can only be used as the default if you can ensure that the interpreter is started before the first and shut down after the last log message.

PR #33235 (2022-01-12): Add capability to redirect to Python logger
**Description of work.**

Defines a new `Poco::Channel` which sends messages to a logger from Python's `logging` framework and `log_to_python` to help set up the logger.

The Python logger has fewer levels than Poco's. It is possible to define new levels in Python but those might clash with user defined levels. Given that the new mechanism is meant to be used in combination with other packages that might log, defining new levels could cause problems. This PR, therefore,  maps several Poco levels onto the same Python level.

It is unclear to me how th [... cut]
closed 2022-01-12 by SimonHeybrock