mantidproject/mantid #33219: Soft Crash Error Report does not send

DanielMurphy22, 2022-01-06:
Found in #33200 

**Describe the bug**
The error reporter, generated on a soft crash (workbench doesn't close), will not send with "Yes, share info". This maybe related to the problem stopping the "Show More details" button in the bottom right of the Error Reporter from working.

**To Reproduce**
- Open Workbench
- Load the file Training_Exercise3a_SNS.nxs from the Trainingcoursedata
- Execute the algorithm NormaliseToMonitor
- The error reporter will appear. Try to click Show more details (a pop-up window should appear, but now does not)
- Click "Yes, Share info". It will fail reporting a bad request 400

**Expected behavior**
-Show more details pop-up works and sending error reports with "Yes, Share Info" works too!

**Screenshots**
[image]
 explain your problem. -->

**Platform/Version (please complete the following information):**
 - OS: IDAaaS Centos7 (also found on nightly on Windows)
 - Mantid Version  6.2.20220103.659

DanielMurphy22, 2022-01-07:
The breaking change is these 3 lines here: https://github.com/mantidproject/mantid/pull/33056/files#diff-bf1a2e80801b922a0520f50d5ab63455cfdfd8b13f06ce5592a140ac42238b8fR49-R51

martyngigg, 2022-01-10:
Fixed by #33243

PR #33243 (2022-01-10): Fix error reporter

**Description of work.**

The error reporter was broken because escape characters were being removed from quotes in json strings, so the json parsing failed. This meant that both:
- stack traces (which normally contain `"`s)
- entering a quotation mark in any other fields  (name, email, decription box)
would result in a problem when either:
- clicking the "Show more details" button, or
- clicking the "Yes, share information" button.

We've added tests to ensure that strings with escaped quotes are preserved when converting to json, as well as removing the seemingly unused c [... cut]
closed 2022-01-10 by martyngigg