mantidproject/mantid #33223: Superplot: trying to overlay MD with Histo causes unhandled exception

DavidFair, 2022-01-06:
**Describe the bug**
Trying to plot a MD workspace over a Histo will result in an unhandled exception in Superplot.
The Superplot widget should prevent the use mixing the two workspace types initially, if this behaviour is not supported. Or check that all workspaces types are compatible instead.

**To Reproduce**
```python
ws = CreateSampleWorkspace()
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[-1,1,-1,1,-1,1,-10,10], Names="H,K,L,E", Units="U,U,U,V") FakeMDEventData(InputWorkspace=md_4D, PeakParams='500000,0,0,0,0,3')
```
- Superplot ws
- Drag md_4d onto the Superplot
- Try clicking md_4d, this will crash

**Expected behavior**
This should be handled gracefully

DannyHindson, 2022-01-06:
Possible separate issue - but trying to add an MDWorkspace onto a Colorfill plot (by dragging the MDWorkspace onto the plot) also creates an unhandled exception:
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\workbench\plotting\figurewindow.py", line 108, in dropEvent
    self._plot_on_here(workspace_names, ax)
  File "C:\MantidNightlyInstall\bin\workbench\plotting\figurewindow.py", line 139, in _plot_on_here
    pcolormesh(names, fig=fig)
  File "C:\MantidNightlyInstall\bin\mantid\plots\plotfunctions.py", line 61, in inner_func
    return func(workspaces, *args, **kwargs)
  File "C:\MantidNightlyInstall\bin\mantidqt\plotting\functions.py", line 244, in pcolormesh
    _validate_pcolormesh_inputs(workspaces)
  File "C:\MantidNightlyInstall\bin\mantidqt\plotting\functions.py", line 324, in _validate_pcolormesh_inputs
    raise_if_not_sequence(workspaces, 'workspaces', MatrixWorkspace)
  File "C:\MantidNightlyInstall\bin\mantid\plots\plotfunctions.py", line 321, in raise_if_not_sequence
    list(map(raise_if_not_type, value))
  File "C:\MantidNightlyInstall\bin\mantid\plots\plotfunctions.py", line 317, in raise_if_not_type
    raise ValueError(f"{x} has unexpected type: '{x.__class__.__name__}'")
ValueError: md_3D has unexpected type: 'MDLeanEventWorkspace3D'

Continue working.
```

gvardany, 2022-01-10:
@gui-co Can you try to handle the exception better? Of course, superplot doesn't support MD yet.

martyngigg, 2022-01-13:
Fixed by #33250

PR #33250 (2022-01-11): Bug - Superplot does not support mdworkspace
**Description of work.**

With this PR, the types of workspaces are checked before being added to the list in Superplot. This will avoid any exception when trying to plot unsupported workspaces.

**To test:**

See issue description.

Fixes #33223 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
 [... cut]
closed 2022-01-13 by martyngigg