mantidproject/mantid #33229: Inconsistent colorbar autoscaling in sliceviewer

RichardWaiteSTFC, 2022-01-07:
**Describe the bug**
Found during manual testing of sliceviewer #33201

There are two autoscaling issues that may be related:

1. MatrixWorkspace: Inconsistent behaviour of autoscaling when zooming and clicking Home - get different numbers after unchecking/rechecking autoscaling
2. For MDWorkspace (2D/3D/Histo) the color scale is not updating as you zoom

**To Reproduce**

To test (1)
1. Load `CNCS_7860_event.nxs` and open in sliceviewer
2. Make a note of the maximum color limit
2. Check autoscsale is on/checked
3. Zoom in - note colorscale will change
4. Click Home - colorscale will change but will not go back to original limit
5. Click Home - will change again, this time back to original limit

To test (2)
(1) make the required workspaces
```
md_4D_1 = CreateMDWorkspace(Dimensions=4, Extents=[-1,1,-2,2,-3,3,-4,4], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D_1, UniformParams='5e6') # 4D data
md_4D_2 = CreateMDWorkspace(Dimensions=4, Extents=[-0.5,0.5,-1,2,-3,2,-4,4], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D_2, UniformParams='5e6') # 4D data
md_4D = md_4D_1 + md_4D_2

mdHisto_3D = BinMD(InputWorkspace=md_4D, AlignedDim0='H,-1,1,100', AlignedDim1='K,-1,1,100', AlignedDim2='L,-1,1,100') # 3D cut
mdHisto_2D = BinMD(InputWorkspace=md_4D, AlignedDim0='H,-1,1,100', AlignedDim1='K,-1,1,100') # 2D cut
```
(2) Open `mdHisto_3D` or `mdHisto_2D` in sliceviewer
(3) Zoom in - colorscale won't change when zooming (it should!)

sf1919, 2022-01-07:
Is this the same as #33214?

RichardWaiteSTFC, 2022-01-07:
> Is this the same as #33214?

Yes 🤦‍♂️ - sorry! I will close this and add some detail to the original issue that Martyn made!

closed 2022-01-07 by RichardWaiteSTFC