mantidproject/mantid #33230: ADS observer/handler not handling workspace deletion correctly

RichardWaiteSTFC, 2022-01-07:
**Describe the bug**
Found during manual testing of sliceviewer #33201

1. Deleting an MD workspace in ADS does not close sliceviewer (unless a peaks table is overlain!) and it is still possible to zoom (i.e. it successfully rebins the workspace even though it no longer exists in ADS).
2. When a peak table is overlaid on sliceviewer and then deleted on the ADS it should remove that table from the peaksviewer (if it is the only table viewed then it should close the peaks viewer). Atm the table is still shown and then double clicking on a peak (in that table or any other is overlain) will cause a "variable invalidated error" exception

**To Reproduce**
To-Do

**Expected behavior**
- Deleting a workspace viewed in sliceveiwer would close sliceviewer
- Deleting a peaks workspace overlain will remove the table from sliceviewer

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

RichardWaiteSTFC, 2022-01-10:
Looks like the work done in this PR #32642 has been removed somehow!

martyngigg, 2022-01-13:
Fixed by #33253

closed 2022-01-13 by martyngigg