mantidproject/mantid #33244: Qt error when Remove selected then Remove all rows in EngDiff UI table

RichardWaiteSTFC, 2022-01-10:
**Describe the bug**
Found during manual testing for 6.3 release #33191 
Qt error when Remove Selected then Remove all rows in EngDiff UI table
```
Error occurred in handler:
Traceback (most recent call last):
File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\fitting_ads_observer.py", line 20, in wrapper
func(*args, **kwargs)
File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\fitting_ads_observer.py", line 75, in replaceHandle
self.replace_callback(name, workspace)
File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_widget.py", line 40, in replace_workspace
self.presenter.replace_workspace(name, workspace)
File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_presenter.py", line 105, in replace_workspace
self._repopulate_table()
File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_presenter.py", line 139, in _repopulate_table
self._remove_all_table_rows()
File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_presenter.py", line 290, in _remove_all_table_rows
self.view.remove_all()
File "C:\MantidNightlyInstall\bin\mantidqtinterfaces\Engineering\gui\engineering_diffraction\tabs\fitting\data_handling\data_view.py", line 189, in remove_all
self.table_selection.setRowCount(0)
RuntimeError: wrapped C/C++ object of type QTableWidget has been deleted
```
**To Reproduce**
Having trouble reproducing...
(1) Load a few (2-3?) focussed spectra into the fitting tab of EngDiff UI
(2) Click a row in the table and click button Remove Selected
(3) Click Remove All button

**Expected behavior**
No exception, table in UI would be cleared and the `_logs` group of workspaces in ADS will be deleted.

**Platform/Version (please complete the following information):**
 - Nightly 6.2 (4 Dec) windows

RichardWaiteSTFC, 2022-01-10:
Could be related #33220

DannyHindson, 2022-01-12:
I can't reproduce this on Windows. I think I have seen this error though occasionally while doing development work on this UI but haven't been able to pin down the actions that cause it

RichardWaiteSTFC, 2022-01-12:
Perhaps the most reproducible way of triggering the same/similar error is in #33281?

martyngigg, 2022-01-14:
Fixed by #33299

PR #33299 (2022-01-14): Various small fixes to the Engineering UI Fitting tab
**Description of work.**

Various related bug fixes that have come out of manual testing for the 6.3 release:

- remove error when renaming a workspace that has been plotted on the fitting tab (#33281). The same code change may also fix #33244 but I've not been able to reproduce that one
- I've occasionally seen the addition of a BackToBackExponential peak generating an error message that simply says "unknown" from a call to the setPeakFwhmOf function. I've updated the sip file to generate a better message. Annoyingly I've not managed to re [... cut]
closed 2022-01-14 by martyngigg