mantidproject/mantid #33249: Crash when trying to change setting in config file

RichardWaiteSTFC, 2022-01-10:
**Describe the bug**
Found during manual testing for 6.3 release - #33191
Strange crash when trying to close the About Mantid popup that seemed to be a problem with the config file - but hard to reproduce. 

**To Reproduce**
(1) Open up EngDiff interface and change a setting (cog in bottom left) - e.g. check the box next to a log in the fitting section of the settings.
(2) Close the UI
(3) Change default facility Help > About mantid > Facility - and close the pop up
(3) Try to change a setting in mantid File > Settings > (at this point it crashes)

```
Traceback (most recent call last):
 File "C:\MantidNightlyInstall\bin\workbench\widgets\about\view.py", line 47, in closeEvent
 self.presenter.save_on_closing()
 File "C:\MantidNightlyInstall\bin\workbench\widgets\about\presenter.py", line 198, in save_on_closing
 self.parent.config_updated()
 File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 701, in config_updated
 self.editor.load_settings_from_config(CONF)
 File "C:\MantidNightlyInstall\bin\workbench\plugins\editor.py", line 59, in load_settings_from_config
 self.editors.load_settings_from_config(config)
 File "C:\MantidNightlyInstall\bin\mantidqt\widgets\codeeditor\multifileinterpreter.py", line 93, in load_settings_from_config
 self.confirm_on_save = config.get('project', 'prompt_save_editor_modified')
 File "C:\MantidNightlyInstall\bin\workbench\config\user.py", line 94, in get
 raise_keyerror(full_option)
 File "C:\MantidNightlyInstall\bin\workbench\config\user.py", line 87, in raise_keyerror
 raise KeyError('Unknown config item requested: "{}"'.format(option))
KeyError: 'Unknown config item requested: "project"'
```

Close/Re-open mantid and it seems to work OK - settings no longer corrupted and can change facility etc. without error

**Expected behavior**
No error...

**Platform/Version (please complete the following information):**
 - Nightly 6.2 (4 Dec) window

DavidFair, 2022-01-19:
I've cannot reproduce this on Linux
I'll retest on Windows

DavidFair, 2022-01-19:
I can't reproduce this on Windows either, I suspect there is either something subtle we are missing. Or a file handle was locked preventing us from re-reading it :thinking: 

I'm going to close, but if anyone managed to reproduce this feel free to re-open with steps

closed 2022-01-19 by DavidFair