mantidproject/mantid #33252: Improve scripting of workspace history for algorithm calls with StoreInADS=False

DannyHindson, 2022-01-10:
If you take a workspace that has been generated by the ISIS Powder diffraction scripts and script the workspace history it doesn't run through properly. This is because of the following code from focus.py:

```
def _divide_one_spectrum_by_spline(spectrum, spline, instrument):
    rebinned_spline = mantid.RebinToWorkspace(WorkspaceToRebin=spline, WorkspaceToMatch=spectrum, StoreInADS=False)
    if instrument.get_instrument_prefix() == "GEM":
        divided = mantid.Divide(LHSWorkspace=spectrum, RHSWorkspace=rebinned_spline, OutputWorkspace=spectrum,
                                StoreInADS=False)
        complete = mantid.ReplaceSpecialValues(InputWorkspace=divided, NaNValue=0, StoreInADS=False)
        # crop based off max between 1000 and 2000 tof as the vanadium peak on Gem will always occur here
        return _crop_spline_to_percent_of_max(rebinned_spline, complete, spectrum, 1000, 2000)

    divided = mantid.Divide(LHSWorkspace=spectrum, RHSWorkspace=rebinned_spline,
                            StoreInADS=False)
    complete = mantid.ReplaceSpecialValues(InputWorkspace=divided, NaNValue=0, OutputWorkspace=spectrum)

    return complete
```
The algorithm calls gets scripted as follows:
```
RebinToWorkspace(WorkspaceToRebin='spline_bank_1_1', WorkspaceToMatch='focused_ws-1', OutputWorkspace='rebinned_spline')
Divide(LHSWorkspace='focused_ws-1', RHSWorkspace='__TMP0000020CAACF86E0', OutputWorkspace='divided')
ReplaceSpecialValues(InputWorkspace='__TMP0000020CAACF41D0', OutputWorkspace='focused_ws-1', NaNValue=0)
```
The __TMP workspace doesn't exist

DannyHindson, 2022-01-10:
The StoreInADS=False doesn't get saved in the workspace history so this may be a problem as well as the actual scripting

DannyHindson, 2022-01-11:
I've made a change to sort this out. Doesn't seem especially urgent to fix in 6.3 though - particularly because the code change seems a bit risky in that it changes the validation on workspace properties so that output workspace properties don't have to have a name. I'll put this through as a PR again main for 6.4 I think

martyngigg, 2022-05-10:
Is it worth us opening this fix for `main` this time?

I'm assuming the change is to store the history whenever it has a name? I'm also wondering if we start to break the rule that names only exist in the ADS. We currently generate the tmp name in the workspace property I think. Maybe we should generate the name in the workspace constructor as just the string of the memory address?

I'll roll this to 6.5S1.

robertapplin, 2024-08-02:
This problem is caused on the `Divide` line. It specifies both `StoreInADS=False` and an `OutputWorkspace` name. This causes project save/recovery to populate any generated script with `__TMP#######`, which never actually exist. The solution is to remove the unrequired `OutputWorkspace` property from the `Divide` line, and then the generated scripts should work

See the contents of this PR for more info https://github.com/mantidproject/mantid/pull/36368

jhaigh0, 2024-08-02:
Looks like `focus.py` is quite different now, I think the equivalent function to the one included in the issue is
```
def _normalize_one_spectrum(single_spectrum_ws, vanadium_ws, instrument):
    rebinned_vanadium = mantid.RebinToWorkspace(WorkspaceToRebin=vanadium_ws, WorkspaceToMatch=single_spectrum_ws, StoreInADS=False)
    divided = mantid.Divide(LHSWorkspace=single_spectrum_ws, RHSWorkspace=rebinned_vanadium, StoreInADS=False)
    if instrument.get_instrument_prefix() == "GEM":
        values_replaced = mantid.ReplaceSpecialValues(InputWorkspace=divided, NaNValue=0, StoreInADS=False)
        # crop based off max between 1000 and 2000 tof as the vanadium peak on Gem will always occur here
        complete = _crop_vanadium_to_percent_of_max(rebinned_vanadium, values_replaced, single_spectrum_ws, 1000, 2000)
    else:
        complete = mantid.ReplaceSpecialValues(
            InputWorkspace=divided,
            NaNValue=0,
            InfinityValue=0.0,
            BigNumberThreshold=1e13,
            SmallNumberThreshold=-1e13,
            OutputWorkspace=single_spectrum_ws,
        )

    if instrument.perform_abs_vanadium_norm():
        complete = _perform_absolute_normalization(rebinned_vanadium, complete)

    return complete
```
which doesn't set the outputworkspace property. I'll try and test to see if this issue has been fixed already

PR #37839 (2024-08-30): storeInADS preserved in algorithm history and recreated correctly by script builder
### Description of work

Previously, scripts in the form
```
my_workspace = Algorithm(..., StoreInADS=False)
result = AnotherAlgorithm(InputWorkspace=my_workspace,...)
```
would be recreated via the ScriptBuilder as
```
Algorithm(OutputWorkspace='my_workspace',...)
AnotherAlgorithm(InputWorkspace='__TMP_<uid>',...)
```
which would not run as `'__TMP_<uid>'` is not a workspace in the ADS.

This PR aims to change this behaviour, so workspace histories can be recreated accurately.

#### Summary of work

- `Algorit [... cut]
closed 2024-08-30 by SilkeSchomann