mantidproject/mantid #33260: Instrument view crash

AndreiSavici, 2022-01-10:
**Describe the bug**
Originally seen in StepScanAnalysis interface: after loading a file, user clicks on `Define` the pixel mask. Mantid crashes. The issue seems to be the python code that is calling the instrument view, in particular the `select_tab` function. One can reproduce this with a simple script as well. This used to work in 6.2.0, does not work in nightly or ORNL newer versions.

**To Reproduce**
In Mantid workbech, in the python interpreter, paste the following code (the file is in the standard test data)

```python
w=Load('CNCS_7860')
from mantidqt.widgets.instrumentview.api import get_instrumentview
instrument_view = get_instrumentview('w')
instrument_view.select_tab(2)
instrument_view.show_view()
```
press Enter.

Note that if you run line by line, one at a time, it seems to be OK :scream: 
Also, commenting out the select tab line will make the script work

**Platform/Version (please complete the following information):**
 - OS: RHEL7, Ubuntu 20.04
 - Mantid Version > 6.2.0

**Additional context**
Any fix in the script should be propagated to [StepScanAnalysys interface](https://github.com/mantidproject/mantid/blob/main/qt/scientific_interfaces/General/StepScan.cpp#L343-L348).

martyngigg, 2022-01-11:
@thomashampson You've been looking at a crash there too. Could it be related?

thomashampson, 2022-01-12:
This problem appears to go away when setting the default value `useThread = false` here:
https://github.com/mantidproject/mantid/blob/cc12a281eb2a83e659bfe03e72b70f91354519dc/qt/widgets/instrumentview/inc/MantidQtWidgets/InstrumentView/InstrumentWidget.h#L120

and is likely related to the instrument viewer crashing in Windows debug mode if it's closed while still loading.

martyngigg, 2022-01-13:
> This problem appears to go away when setting the default value `useThread = false` here: [...]

I'm highly supportive of us setting this to false by default as it doesn't sound like it's ready for production in threaded-mode yet?

thomashampson, 2022-01-13:
Debugging in Windows, running the script gives this message box (this might be information that you're missing on Linux systems):
[image]

AndreiSavici, 2022-01-19:
@thomashampson @martyngigg See if the fix works for you as well

martyngigg, 2022-01-20:
Fixed by #33349

PR #33349 (2022-01-19): [SEPC125]  Fix qt thread issue related to instrument view
**Description of work.**

- Original Gitlab Defect: [SPEC125](https://code.ornl.gov/sns-hfir-scse/spectroscopy/spectroscopy/-/issues/125)

This PR introduces a patch to resolve a Qt thread issue related to instrument viewer.
The issue is described in #33260 , and the cause of the related segfault is due to the incomplete construction of the instrument viewer.

**To test:**

On an analysis cluster (for the test data), run the follow script:

```
from mantidqt.widgets.instrumentview.api import get_instrumentview

w = Load("CNCS_7860")
vie [... cut]
closed 2022-01-20 by martyngigg