mantidproject/mantid #33262: Save docs to PDF Crash and close

DanielMurphy22, 2022-01-11:
Found in #33261 
**Describe the bug**
Try to print a page from the built-in docs to PDF -> Crash and close

**To Reproduce**
Open Workbench
Go to Help > Mantid Concepts
Click Print...
Try to Save as PDF
Crash and Close
PDF not saved

<details>
<summary>Click to see MacOS crash report: </summary>
<br>
Process:               Python [3744]
Path:                  /Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.mantidproject.MantidWorkbenchNightly
Version:               6.2.20220104.1619 (6.2.20220104.1619)
Code Type:             X86-64 (Native)
Parent Process:        MantidWorkbenchNightly [3743]
Responsible:           MantidWorkbenchNightly [3743]
User ID:               501

Date/Time:             2022-01-11 09:37:07.735 +0000
OS Version:            macOS 11.2.3 (20D91)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        76A757C0-05CA-0881-303D-D9D048503406

Sleep/Wake UUID:       E9EC01FF-2FB9-450A-9A69-E216EE943047

Time Awake Since Boot: 49000 seconds
Time Since Wake:       1500 seconds

System Integrity Protection: disabled

Crashed Thread:        35  QThread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000007391390c
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [3744]

VM Regions Near 0x7391390c:
--> 
    __TEXT                      10741c000-10741e000    [    8K] r-x/rwx SM=COW  /Applications/MantidWorkbenchNightly.app/Contents/Frameworks/Python.framework/Versions/3.8/Resources/Python.app/Contents/MacOS/Python

Thread 0:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff2061ce7e mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff2061d1f0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff20749bf7 __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff207482ca __CFRunLoopRun + 1315
4   com.apple.CoreFoundation      	0x00007fff207476ce CFRunLoopRunSpecific + 563
5   com.apple.HIToolbox           	0x00007fff289cf630 RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff289cf42c ReceiveNextEventCommon + 709
7   com.apple.HIToolbox           	0x00007fff289cf14f _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff22f679b1 _DPSNextEvent + 883
9   com.apple.AppKit              	0x00007fff22f66177 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1366
10  com.apple.AppKit              	0x00007fff22f5868a -[NSApplication run] + 586
11  libqcocoa.dylib               	0x000000012e17c5e3 0x12e144000 + 230883
12  org.qt-project.QtCore         	0x000000011c342adf QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
13  org.qt-project.QtCore         	0x000000011c347062 QCoreApplication::exec() + 130
14  QtWidgets.so                  	0x0000000122e03750 meth_QApplication_exec_(_object*, _object*) + 80
15  Python                        	0x000000010743b01d cfunction_call_varargs + 320
16  Python                        	0x000000010743aa73 _PyObject_MakeTpCall + 266
17  Python                        	0x00000001074db49f call_function + 820
[... 1223 lines cut ...]
    0x7fff3f961000 -     0x7fff3fa4efff  com.apple.PDFKit (1.0 - 950.20.2) <5261E71F-D2F2-3CEA-AEF5-F3A817164DBF> /System/Library/Frameworks/PDFKit.framework/Versions/A/PDFKit
    0x7fff40773000 -     0x7fff407f6fff  com.apple.TextInput (1.0 - 1.0) <858479E0-C5AA-30AD-A630-48FC82A5D1F3> /System/Library/PrivateFrameworks/TextInput.framework/Versions/A/TextInput
    0x7fff417dc000 -     0x7fff4180bfff  libncurses.5.4.dylib (57) <CAD79205-84C0-3731-AED8-EDEF51ACC872> /usr/lib/libncurses.5.4.dylib
    0x7fff4180c000 -     0x7fff41815fff  com.apple.IOAccelMemoryInfo (1.0 - 1) <544B95BE-810E-3D94-AEB1-D5C117D946D7> /System/Library/PrivateFrameworks/IOAccelMemoryInfo.framework/Versions/A/IOAccelMemoryInfo
    0x7fff4186d000 -     0x7fff41870fff  com.apple.quartzframework (1.5 - 25) <19311613-BABD-3AF9-A347-06D13169B4EF> /System/Library/Frameworks/Quartz.framework/Versions/A/Quartz
    0x7fff41c4c000 -     0x7fff41c6afff  libCGInterfaces.dylib (544.2) <2BC461CE-B7FA-33F3-A36B-64F54FB371B5> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vImage.framework/Versions/A/Libraries/libCGInterfaces.dylib
    0x7fff441e3000 -     0x7fff44238fff  com.apple.DifferentialPrivacy (1.0 - 1) <7DFAF0E9-86EA-31D6-B34F-4E7F6DC8407E> /System/Library/PrivateFrameworks/DifferentialPrivacy.framework/Versions/A/DifferentialPrivacy
    0x7fff44239000 -     0x7fff44252fff  com.apple.SafariFoundation (16610 - 16610.4.3.1.7) <70C11F1E-353B-30AE-A272-1D778EAC6ED7> /System/Library/PrivateFrameworks/SafariFoundation.framework/Versions/A/SafariFoundation
    0x7fff44253000 -     0x7fff44259fff  com.apple.SafariPlatformSupport.framework (16610 - ???) <8EC01462-CBB4-35EF-A083-2B631B52E034> /System/Library/PrivateFrameworks/SafariPlatformSupport.framework/Versions/A/SafariPlatformSupport
    0x7fff4425a000 -     0x7fff444d3fff  com.apple.RawCamera.bundle (9.10.0 - 1430) <3E7403EB-156C-36C7-93AD-1356447D5310> /System/Library/CoreServices/RawCamera.bundle/Contents/MacOS/RawCamera
    0x7fff44e23000 -     0x7fff44e27fff  com.apple.DAAPKit (1.0 - 1) <B61034CC-045C-3008-B21B-9E0B7FC70F98> /System/Library/PrivateFrameworks/DAAPKit.framework/Versions/A/DAAPKit
    0x7fff46f5b000 -     0x7fff46f66fff  libGPUSupportMercury.dylib (18.1.1) <1321060E-8F56-3917-8F4D-510A4DB45798> /System/Library/PrivateFrameworks/GPUSupport.framework/Versions/A/Libraries/libGPUSupportMercury.dylib
    0x7fff485c9000 -     0x7fff48618fff  com.apple.osanalytics.OSAnalytics (1.0 - 1) <CADD88A2-87A8-3879-8759-D3C112A6844F> /System/Library/PrivateFrameworks/OSAnalytics.framework/Versions/A/OSAnalytics
    0x7fff52557000 -     0x7fff52585fff  com.apple.HelpData (2.3 - 200) <76AA2951-C5AA-3A31-A4CA-62E28DF02A84> /System/Library/PrivateFrameworks/HelpData.framework/Versions/A/HelpData
    0x7fff52586000 -     0x7fff525a0fff  com.apple.shortcut (2.16 - 117) <389381A8-F3EF-3143-AE06-1D91D0D57777> /System/Library/PrivateFrameworks/Shortcut.framework/Versions/A/Shortcut
    0x7fff525a1000 -     0x7fff52633fff  com.apple.Symbolication (12.2 - 64541.7.1) <FA39B702-6E08-3F11-953F-5EC1EFB993BB> /System/Library/PrivateFrameworks/Symbolication.framework/Versions/A/Symbolication
    0x7fff54ea7000 -     0x7fff54ebcfff  com.apple.ScriptingBridge (1.4 - 80) <ADC81D20-F770-3983-B7AA-A7DA2B428919> /System/Library/Frameworks/ScriptingBridge.framework/Versions/A/ScriptingBridge
    0x7fff5511c000 -     0x7fff55153fff  com.apple.ReplayKit (1.0 - 1) <756C96A4-B07B-3780-B75E-29FB79336195> /System/Library/Frameworks/ReplayKit.framework/Versions/A/ReplayKit
    0x7fff56777000 -     0x7fff5677afff  com.apple.ForceFeedback (1.0.6 - 1.0.6) <8782199E-5556-3FB6-87E8-03472B480BAA> /System/Library/Frameworks/ForceFeedback.framework/Versions/A/ForceFeedback
    0x7fff5ed7c000 -     0x7fff5f20cfff  com.apple.driver.AppleIntelKBLGraphicsMTLDriver (16.1.12 - 16.0.1) <2B7EEE40-F4BD-32CD-B5A3-51F67B57AA69> /System/Library/Extensions/AppleIntelKBLGraphicsMTLDriver.bundle/Contents/MacOS/AppleIntelKBLGraphicsMTLDriver
    0x7fff5f20d000 -     0x7fff5fc20fff  libSC.dylib (4.2.15) <86667FD2-E9BB-39FE-82E9-7B1F0230A922> /System/Library/Extensions/AMDShared.bundle/Contents/PlugIns/libSC.dylib
    0x7fff5ff69000 -     0x7fff60f4cfff  com.apple.driver.AppleIntelKBLGraphicsGLDriver (16.1.12 - 16.0.1) <7E0C9E81-E41B-3D24-AB0B-84295B61FADF> /System/Library/Extensions/AppleIntelKBLGraphicsGLDriver.bundle/Contents/MacOS/AppleIntelKBLGraphicsGLDriver
    0x7fff6a732000 -     0x7fff6a736fff  libmetal_timestamp.dylib (31001.161) <DF0A6F36-E486-3656-A6F2-9BA9A0CF2159> /System/Library/PrivateFrameworks/GPUCompiler.framework/Versions/31001/Libraries/libmetal_timestamp.dylib
    0x7fff6cba0000 -     0x7fff6cba6fff  libCoreFSCache.dylib (177.22) <44489BD1-3963-35DF-86F1-DE95778AC0DB> /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libCoreFSCache.dylib
    0x7fff6cba7000 -     0x7fff6cbabfff  libCoreVMClient.dylib (177.22) <989F6CBF-CCEF-340D-9CB5-EC4133707040> /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libCoreVMClient.dylib

DanielMurphy22, 2022-01-12:
I can reproduce this on MacOS BigSur and Windows 10, NOT Ubuntu 18.04. It seems to be an issue in 6.0.0 so this has been around a few releases and is not a priority.

DanielMurphy22, 2022-01-12:
Worth looking at mantid/qt/widgets/common/src/pqHelpWindow.cxx printPage() L373

martyngigg, 2022-05-05:
It's worth retesting this with the Conda produced packages so I'll keep this at 6.4

martyngigg, 2022-05-12:
I'm going to try this with the new package.

martyngigg, 2022-05-17:
Confirmed that this is still and issue in the conda-packaged version.

martyngigg, 2022-06-08:
I've tried putting breakpoints in `pqHelpWindow::printPage` and it still crashes before testing `dialog.exec`. The latest error I saw was:
```
022-06-08 12:01:09.634615+0100 python[45712:2599153] [static] Failed to realize static UTType instance 0x6000005586c0 for identifier pdf. The faulting component was unknown at runtime.
```
I think this could be an internal Qt error. We'll be upgrading to 5.15 after the release. Let's re-evaluate then.

martyngigg, 2023-01-24:
Closing as this is fixed by #35043

closed 2023-01-24 by martyngigg