mantidproject/mantid #33266: IDA fit tab GUI can be disrupted if a fit produces a chisq too large

Harrietbrown, 2022-01-11:
This is difficult to replicate, but in a case where a fit in IDA has a large chisq, meaning has a large number of digits, it pushes the gui to the side and cannot be fixed without closing and re opening. this simply requires the string passed into the label be formatted is the value is >~10000

AnthonyLim23, 2022-01-12:
iris26176_graphite002_QLr_Result_HWHM and doing FickDiffusion with D fixed to a value of 1e6, causes a huge chi squared of 35826081236140572672

However, I dont have problems viewing the GUI. Maybe this only happens on laptops?

Harrietbrown, 2022-01-13:
It happened on IDASS during testing. it was much larger than 35826081236140572672, enough digets that is covered the entire gui

AnthonyLim23, 2022-01-27:
I have been trying to cause a bad fit and the chi-squared is still too low to see this issue. Since I am going out of my way to get a bad fit and it is still not "bad enough" to cause the bug, I think it is safe to say a user (who will be trying to get a good fit) should never see this.

Harrietbrown, 2022-01-28:
I am in aggreement here, I have yet to be able to replicate the issue propperly, so I am goign to close this for now.

closed 2022-01-28 by Harrietbrown