mantidproject/mantid #33295: Unhandled Ex When Changing Plot and SubBG Options With Fit Window Open

ConorMFinn, 2022-01-12:
**Describe the bug**
Mantid throws when either of these checkbox options are changed with the fit browser open. 

**To Reproduce**
1. Open the Eng Diff GUI and navigate to the fitting tab
2. Load some focused data (see https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html for instructions)
3. Plot the data
4. Click the fit button to open the fit browser
5. Check/Uncheck either of plot or sub-bg

**Expected behavior**
No errors

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: Ubuntu, not windows
 - OS Version: 18.04
 - Mantid Version 6.3

DavidFair, 2022-01-20:
I've tested this locally with a different / the same file as Conor and cannot reproduce it

ConorMFinn, 2022-01-20:
Unable to be reproduced on two other systems, both windows and another ubuntu machine. Seems to be some weird quirk of my system.

closed 2022-01-20 by ConorMFinn