mantidproject/mantid #33310: ROI Preview error on SliceViewer

sf1919, 2022-01-14:
**Describe the bug**
When testing #33297 I found that pressing the ROI Preview button caused an error message to pop up and the following to appear in the message window - `WARNING:matplotlib.text:posx and posy should be finite values`. The lines plots don't show anything at all.

This issue also occurred on the most recently Nightly.

**To Reproduce**
1. Open Mantid
2. Run the following script
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-6,6,-4,4,-2,2',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                       Frames='HKL,HKL,HKL',
                       SplitInto='2', SplitThreshold='50')
expt_info = CreateSampleWorkspace()
ws.addExperimentInfo(expt_info)
SetUB(ws, 1,1,2,90,90,120)
FakeMDEventData(ws, EllipsoidParams='1e+04,0,1,1,1,0,0,0,1,0,0,0,1,0.02,0.06,0.03,1', RandomSeed='3873875')
mdHisto_2D = BinMD(InputWorkspace=ws, 
    AlignedDim0='H,-2,2,100', AlignedDim1='K,-2,2,100') # 2D cut
```
3. Open `mdHisto_2D` in sliceviewer
4. Click on the ROI Preview button
5. An error message pops up

**Expected behavior**
I expected to press the ROI Preview button and no error messages occur. I also expect the line plots to work.

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
Windows 10
Mantid Nightly 6.2.20220113.1836

RichardWaiteSTFC, 2022-01-14:
Thanks for making this issue. I can't reproduce this - maybe someone else could have a go @DanielMurphy22?
As well as the error does the roi tool not work, and no 1D curves plotted?

DanielMurphy22, 2022-01-14:
I can't reproduce on Ubuntu, trying to build on MacOS

RichardWaiteSTFC, 2022-01-14:
Can only reproduce when colorscale is power (as opposed to linear or log) - affects line plots as well as ROI (i.e. is an issue with line plots)

DanielMurphy22, 2022-01-14:
I couldn't reproduce with on MacOS or Windows. I've seen that sort of Matplotlib error before though. Possibly it's something that you have to get Mantid or python into a certain state for the error to show.

Sorry to distract the conversation, but I do get this error on the nightly after selecting an ROI and then trying to change the axes that H and K apply to. (On Windows10 nightly13thJan2022):
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\sliceviewer\presenter.py", line 207, in dimensions_changed
    if sliceinfo.slicepoint[data_view.dimensions.get_previous_states().index(None)] is None:
ValueError: None is not in list
```

DanielMurphy22, 2022-01-14:
Changing **Colorbar to Power** is the key! I can reproduce it now! Well Done Richard!

RichardWaiteSTFC, 2022-01-14:
> I couldn't reproduce with on MacOS or Windows. I've seen that sort of Matplotlib error before though. Possibly it's something that you have to get Mantid or python into a certain state for the error [...]

That is fixed by #33297, the PR Sarah was testing. We've gone full circle!

RichardWaiteSTFC, 2022-01-14:
> Changing **Colorbar to Power** is the key! I can reproduce it now! Well Done Richard!

Thanks to @sf1919  for thorough testing!

sf1919, 2022-05-24:
Fixed by #33941

closed 2022-05-24 by sf1919