mantidproject/mantid #33337: Move GSL to 2.7 for conda-forge compatibility

Pasarus, 2022-01-18:
**Describe the outcome that is desired.**

Conda-forge has a bunch of globally pinned packages, with that in mind, there is a general effort in the community to ensure these packages are used for as many other packages as possible.

**Does the outcome relate directly to a problem? Please describe.**

Ensuring compatibility with conda-forge for the upcoming release.

**Describe any solutions you are considering**

Move to GSL 2.7 in conda envs.

**Additional context**

Conda forge global pinning: https://github.com/conda-forge/conda-forge-pinning-feedstock/blob/master/recipe/conda_build_config.yaml

DavidFair, 2022-01-20:
Can we push this to 6.4?
I'm not sure we want to update a dependency so late into testing, it will require us to retest fitting components and the GUIs associated as they heavily use GSL

Pasarus, 2022-01-20:
The thing is, we are deploying on Conda with gsl 2.7 by sticking with the conda-forge compatibility pinning. We have 2 options, deploy on conda with a different version and hope for the best, or pin back and reduce mantid compatibility with other software on conda-forge.

DavidFair, 2022-01-20:
Lets leave Conda forge on 2.7 (since it's not a fully supported platform of this release, and we can always do a conda only patch)

Then move main onto 2.7 to bring us in-line for the next release? Also it would be good to check if any other packages need updating similarly at the same time

martyngigg, 2022-05-03:
So we lost track of this because the EPIC got closed but this in particular out of that epic is worth doing.

PR #34291 (2022-08-18): Update GSL to v2.7
**Description of work.**
Update the conda yml files to move Mantid to GSL v2.7.

**To test:**
1. Check this passes on all tests
2. Code review

Fixes #33337 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests small and test the class in isolation?
- If there is GUI work does it follow the [GUI standards [... cut]
closed 2022-08-18 by martyngigg