mantidproject/mantid #33338: Indirect data reduction Symmetrise tab can raise errors with setting input values

Harrietbrown, 2022-01-18:
**Original reporter:** Sanghamitra

**Describe the bug**
In the Indirect Data Reduction Symmetrise tab it is possible to arange the input parameters from the plot to raise errors from the symeterise algorithm

**To Reproduce**
1. Load `osi100324_graphite002_red` from sample data.
2. open Indirect Data Reduction and switch to the Symmetrise tab.
3. switch Input from file to Workspace, select osi100324_graphite002_red fromt he dropdown
4. in the Symmetrise plot window there should be 4 vertical green lines, drag the far right hand line to 2, the far left line will vanish (it is set to zero and overlaps with another line)
5. drag the middle line to the right, two green lines should appear in the negative area.
6. click preview. this raises an error without valuble feedback. 

**Expected behavior**
The UI should do two things, make best efforts to prevent invalid input paramiters from getting to the algorithm, and if values do get to the algoithm the algoirthm validation should give relivant error messages at the validation stage.

Ensuring a valid error message will me handled in a PR for 6.3 release, fixing the UI to prevent invalid inout will be an issue for 6.4

AnthonyLim23, 2022-04-06:
Has this been fixed?

SpencerHowells, 2022-04-06:
@AnthonyLim23 Not that I know of. Running it has always been a bit difficult/complicated.