mantidproject/mantid #33340: Fix inconsistent scaling of errors on fit curves by reduced chi-squared

RichardWaiteSTFC, 2022-01-18:
**Describe the bug**

Mantid currently does not scale the covariance matrix from the fit by the reduced chi-squared when determining the errors on the best fit parameters (note this is different from the default scipy `curve_fit`/`leastsq` behaviour - see absolute_sigma=False in the scipy docs: https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.curve_fit.html). However mantid does scale the covariance matrix by the reduced chi-squared when evaluating the errors on the fitted curve).
https://docs.mantidproject.org/v6.1.0/algorithms/Fit-v1.html#overview

Scaling the covariance matrix by the recued chi-squared is equivalent to scaling the errorbars on the data points to force reduced chi-squared == 1 - this is not the desired default behaviour for mantid fitting.

This issue has the following aims:

1. Stop scaling of the covariance matrix by the reduced chi-squared when evaluating the error on the fit curve
2. Update Fit algorithms docs to say change in behaviour for 6.3 (new changelog section)
3. Add a section in the docs to explain difference in behaviour scipy/mantid - they are consistent when `absolute_sigma=True` in curve_fit

**To Reproduce**
(1) Run this script
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
from scipy import optimize
from scipy.stats import t

def func(x, c ,m):
    return m*x + c
    
def confidence_band(x, yfit, covar, dfdp, confprob=0.6826):
   alpha = 1 - confprob
   prb = 1.0 - alpha/2  # +/- half this
   nparams = pcov_true.shape[0]
   dof = len(x)-nparams
   tval = t.ppf(prb, dof)
   df_sq = np.zeros(x.shape)
   for j in range(nparams):
      for k in range(nparams):
         df_sq  += dfdp[j]*dfdp[k]*covar[j,k]
   df = np.sqrt(df_sq)
   delta = tval * df   
   upperband = yfit + delta
   lowerband = yfit - delta 
   return upperband, lowerband

# Generate data
noise_stdev = 0.1
err_scale = 0.5  # 1 => chi-squared = 1 (i.e. errorbars will correspond to stdev of gauss noise)
m0 = 1
c0 = 0.5
x = np.linspace(0,1,20)
np.random.seed(1)
y = np.random.normal(func(x, c0, m0), noise_stdev)
e = err_scale*noise_stdev*np.ones(x.shape)
ws = CreateWorkspace(DataX=x,DataY=y, DataE=e, EnableLogging=False)

# initial guess
p0 = [1,2]  # c, m

# scipy
popt, pcov_false = optimize.curve_fit(func, x, y, p0, sigma=e,
    absolute_sigma=False)
perr_false = np.sqrt(np.diag(pcov_false))
_, pcov_true = optimize.curve_fit(func, x, y, p0, sigma=e, 
    absolute_sigma=True)  # popt does not depend on absolute_sigma
perr_true = np.sqrt(np.diag(pcov_true))
chisq = np.sum(((y - func(x, *popt))/e)**2)
red_chisq = chisq/(len(x) - len(popt))
# calc confidence limits
dfdp = [1, x]  # d(func)/dparam = [dy/dc, dy/dm] = [1, x]
upper_true, lower_true = confidence_band(x, func(x, *popt), pcov_true, dfdp)
upper_false, lower_false = confidence_band(x, func(x, *popt), pcov_false, dfdp)

# mantid
fit = Fit(Function='name=LinearBackground,A0={0},A1={1}'.format(*p0), 
    InputWorkspace='ws', IgnoreInvalidData=True, Output='ws', OutputCompositeMembers=True)

popt_mtd = [fit.Function.function.getParameterValue(iparam) for 
    iparam in range(fit.Function.function.nParams())]
perr_mtd = [fit.Function.function.getError(iparam) for 
    iparam in range(fit.Function.function.nParams())]
red_chisq_mtd = fit.OutputChi2overDoF

# compare mantid and scipy

print('### scipy curve_fit ###')
print("red chisq = ", red_chisq)
print("popt = ", popt)
print('abs_sigma=False:\t perr = ', perr_false)
print('abs_sigma=True:\t perr = ', perr_true)
print('### mantid ###')
print("red chisq = ", red_chisq_mtd)
print("popt = ", popt_mtd)
print('perr =', perr_mtd)

fig, ax = plt.subplots(subplot_kw={'projection': 'mantid'})
ax.errorbar(x, y, yerr=e, color = 'k', 
    marker='o', markersize=3, capsize=2, ls='', label = 'data')
ax.plot(ax.get_xlim(), func(np.array(ax.get_xlim()), c0, m0), '-k', label = 'pdf')
ax.plot(x, func(x, *popt), '-b', label = 'scipy curve_fit')
ax.plot(x, upper_true, ':b', label = '68.3% bound (scipy cov abs_sig=True)')
ax.plot(x, lower_true, ':b')
ax.plot(x, upper_false, ':c', label = '68.3% bound (scipy cov abs_sig=False)')
ax.plot(x, lower_false, ':c')
ax.errorbar(fit.OutputWorkspace, wkspIndex=1, 
    color = 'r', ls ='--', capsize=2, label='mtd fit')
ax.set_xlabel('x')
ax.set_ylabel('y = mx + c')
ax.legend(fontsize=8.0).set_draggable(True).legend
fig.show()
```

This produces
```
### scipy curve_fit ###
red chisq =  5.365685134520836
popt =  [0.4945525  0.98422208]
abs_sigma=False:	 perr =  [0.04991203 0.08533464]
abs_sigma=True:	 perr =  [0.02154729 0.03683942]
### mantid ###
red chisq =  5.36568513452084
popt =  [0.49455249691092085, 0.9842220788860124]
perr = [0.021547290184283356, 0.03683941988065035]
```

[image]

**Expected behavior**
Errors on best fit parameters and on fit curve would be consistent

RichardWaiteSTFC, 2022-01-18:
Note the correct definition of the error on the fitted curve from [1] is
[image]
The equation used in mantid is 
[image]
Which applies only to unit or relative weighting (i.e. absolute_sigma=False)
[1] https://www.astro.rug.nl/software/kapteyn/kmpfittutorial.html

martyngigg, 2022-02-01:
Fixed by #33341

PR #33341 (2022-02-01): Fix inconsistent scaling of errors on fit curves by reduced chi-squared
**Description of work.**
To see what breaks for now...
See issue for details

**To test:**
(1) Run the script in the original issue you should see the plot on the RHS (i.e. the error on the fit curve should agree with the 68.3% confidence bounds calculated from the covariance matrix from scipy `curve_fit `with `absolute_sigma=True`
[image]

Fixes #33340 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

-  [... cut]
closed 2022-02-01 by martyngigg