mantidproject/mantid #33376: Indirect Corrections has no checks on inputs having different kinds of data

Harrietbrown, 2022-01-25:
Indirect Correction interface can be made to crash if the sample and container are different kinds of data.

Reported by Sanghamitra Mukhopadhyay

**To Reproduce**
1. open Indirect->Corrections
2. Load assocaited `_sqw` **will be added when obtained**
3. Load associated `_red` **will be added when obtained**

SpencerHowells, 2022-01-25:
@Harrietbrown what corrections tab?

AnthonyLim23, 2022-01-25:
I got this to crash on nightly using: 

osiris126202_multi_graphite002_red.nxs as the sample
osiris126202_multi_graphite002_sqw.nxs as the container 

However, the next attempt threw an error.

sf1919, 2022-01-25:
I'm unable to recreate a crash. If I open the Indirect->Corrections GUI and on the first tab (Container Subtraction) I added a _sqw sample and a _red container file. Nothing happens until I press run at which point the user receives a pop up error explaining the Sample and Container do not have a matching number of histograms

[image]

AnthonyLim23, 2022-01-25:
I think its just an error where something hangs for a little to long at random and this causes a crash

sf1919, 2022-01-25:
I've also now tried it the same way around as @AnthonyLim23 and get the error message but no crash. I'm doing this on a new branch created from release-next

AnthonyLim23, 2022-01-25:
I think its caused at random and we are not going to be able to reliably reproduce it

Harrietbrown, 2022-01-25:
If we are unable to replicate it then this should be a low priority issue

SpencerHowells, 2022-01-25:
@AnthonyLim23 @sf1919 @Harrietbrown In the tabs there is a filter for _red pseudoextension. It is there for a reason!!!! Anyone changing it to _sqw can expect things to go wrong!

AnthonyLim23, 2022-01-27:
I will try again this afternoon to reproduce this, if I cannot then I will close the issue.

AnthonyLim23, 2022-01-27:
I couldnt reproduce this, so I am closing it

closed 2022-01-27 by AnthonyLim23