mantidproject/mantid #33384: Mantid crashes when running an absorbption correction in Indirect

sf1919, 2022-01-26:
**Original reporter:** Spencer Howells

**Describe the bug**
In the Apply Absorption Corrections tab in the Indirect Corrections GUI, pressing the run button after loading a sample and corrections makes Mantid disappear (when using Windows). When using Ubuntu the same action causes a segmentation fault.

**To Reproduce**
1. Open Indirect -> Corrections
2. Open the Apply Absorptions Corrections Tab
3. Load a sample file
4. Load a corrections file
5. Press Run
6. Windows users will see Mantid close, Ubuntu users will get a segmentation fault.

**Expected behavior**
Pressing the run button should not crash Mantid or cause a segmentation fault.

**Platform/Version (please complete the following information):**
Tested by user on Windows and Ubuntu (versions not supplied)

Tested by @sf1919 on Windows 10 using Mantid v6.2.20220124.1600

sf1919, 2022-01-27:
In discussion with user it was suggested that the lack of a check for the number of histograms in the sample and corrections being equal was the cause of the issue. This was not the case and the issue lies elsewhere.

martyngigg, 2022-01-28:
Fixed by #33395

PR #33395 (2022-01-27): Fixed crash in Absorption Corrections Tab in Indirect
**Description of work.**
This PR fixes the specific issue of Mantid closing without error in Windows or a seg fault occuring in Ubuntu when using the Apply Absorptions Corrections Tab. This fix does not include checking histograms as dicsussed in #33384. This will be dealt with in a separate PR as this isn't the cause of the specific crash that is occurring.

**Report to:** Spencer Howells

**To test:**

1. Open Indirect -> Corrections
2. Open the Apply Absorptions Corrections Tab
3. Load a sample file (ask for file if you don't have anythi [... cut]
PR #33403 (2022-01-28): Add a Historgram validation check to Indirect Absorption Corrections
**Description of work.**
In the process of trying to fix #33384 it was suggested that this might be resolved by adding a check to the number of histograms in the sample and corrections workspaces (they need to be the same). This check is already used in other tabs in the Indirect Corrections GUI.

Whilst this did not fix the fault that occurred the addition of this check is sensible and has been done using this PR.

**Report to:** Spencer Howells

**To test:**
1. Using #33395 make changes to lines 183 and 189 of the ApplyAbso [... cut]
closed 2022-01-28 by martyngigg