mantidproject/mantid #33388: Update from ... import statements in ISIS_Powder units

DannyHindson, 2022-01-26:
There are various from ... import statements in the ISIS_Powder units that rely on the .pth file having been processed so that all of the scripts subdirectories are present on the sys path. These statements generate red highlights in PyCharm because this only happens at run whenever the mantid.simpleapi is run:

[image]

This means there are various distracting errors highlighted by PyCharm and it also means that if you have a unit that doesn't import mantid.simpleapi the from ... import statement will fail (even though the same statement works at run time in another unit)

If the module paths were extended with a Diffraction prefix (eg from Diffraction.isis_powder import X) I think the PyCharm warnings would go away

DannyHindson, 2023-01-09:
Keep open

sf1919, 2023-07-20:
Keep open

sf1919, 2024-01-17:
Keep open

sf1919, 2024-07-16:
Keep open