mantidproject/mantid #33389: Unexpected exception in non-orthog view of sliceviewer when color axis autoscaled

RichardWaiteSTFC, 2022-01-26:
**Describe the bug**
Regression introduced in #33336 - enabling non-orthog view in sliceviewer produces an exception realting to the updating 
```
Traceback (most recent call last):
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\sliceviewer\view.py", line 425, in on_non_orthogonal_axes_toggle
    self.presenter.nonorthogonal_axes(state)
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\sliceviewer\presenter.py", line 365, in nonorthogonal_axes
    self.show_all_data_requested()
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\sliceviewer\presenter.py", line 239, in show_all_data_requested
    self.set_axes_limits(*limits)
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\sliceviewer\presenter.py", line 256, in set_axes_limits
    self.data_limits_changed()
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\sliceviewer\presenter.py", line 226, in data_limits_changed
    self.new_plot()  # automatically uses current display limits
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\sliceviewer\presenter.py", line 121, in new_plot_MDE
    data_view.plot_MDH(
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\sliceviewer\view.py", line 296, in plot_MDH_nonorthogonal
    self.draw_plot()
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\sliceviewer\view.py", line 352, in draw_plot
    self.colorbar.update_clim()
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\colorbar\colorbar.py", line 264, in update_clim
    self._autoscale_clim()
  File "c:\mantid\qt\python\mantidqt\mantidqt\widgets\colorbar\colorbar.py", line 298, in _autoscale_clim
    data = self.colorbar.mappable.get_array_clipped_to_bounds()
AttributeError: 'QuadMesh' object has no attribute 'get_array_clipped_to_bounds'
```

**To Reproduce**
(1) Create a workspace
```
md_3D = CreateMDWorkspace(Dimensions=3, Extents=[-0.5,0.5,-1,1,-2,2], 
    Names="H,K,L", Frames='HKL,HKL,HKL',Units='r.l.u.,r.l.u.,r.l.u.')
expt_info = CreateSampleWorkspace()
md_3D.addExperimentInfo(expt_info)
SetUB(Workspace=md_3D, c=2, gamma=120)
```
(2) Open `md_3D` in sliceviewer
(3) Click non-orthog view

**Expected behavior**
No exception - color axis auto scale would work

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

RichardWaiteSTFC, 2022-01-26:
Happens with MDHisto workspace too because `new_plot_MDH `and `new_plot_MDE` end up calling `plot_MDH_nonorthogonal` - for example repeat steps 2-4 with the `md_3D_svrebinned` workspace produced.

martyngigg, 2022-02-01:
Fixed by #33397