mantidproject/mantid #33409: Bug in autoscale color axis for nonortho view of MD(Event)Workspaces in sliceviewer

RichardWaiteSTFC, 2022-01-31:
**Describe the bug**
Color axis autoscaling is working for MDHisto workspaces in nonorthog view in sliceviewer.

Fix for MDEvent workspaces for the 6.3 release was done in #33397, but MDHisto workspaces need to have the color limits determined from the data within the axes extents - see that PR for more info.

**To Reproduce**
Follow the instructions in #33397 - steps 6-7 demonstrates the bug for MDHisto

**Expected behavior**
Autoscaling would work for MDHisto in nonrothog view

RichardWaiteSTFC, 2022-02-02:
To avoid duplicating pcolormesh could just add add `get_array_clipped_to_bounds` as a method `QuadMesh` like so using setattr

```
class A:
    def __init__(self, aval):
        self.aval = aval
        
    def get_aval(self):
        return self.aval

class AA(A): pass

def addmethod(obj, name, func):
    cls = obj.__class__
    # subclass = type(cls.__name__, (cls,), {})  # doesn't change the name
    subclass = type("AA", (cls,), {})
    setattr(subclass, name, func)
    obj.__class__ = subclass
    
def double_aval(self):
    self.aval *=2
    return self.get_aval()
    
a = A(1)
print(type(a), a.get_aval())    # <class 'mantidqt.widgets.codeeditor.execution.A'> 1
addmethod(a, "double_aval", double_aval)
print(type(a), a.double_aval()) # <class 'mantidqt.widgets.codeeditor.execution.AA'> 2
```

RichardWaiteSTFC, 2022-02-03:
Or can do this
```
class QuadMeshMtd(QuadMesh):
    def get_array_clipped_to_bounds():
        return "pink"

x = np.linspace(-1, 1, 12)
y = np.linspace(-2, 2, 20)
xx, yy = np.meshgrid(x, y)

fig, ax = plt.subplots()
mesh = ax.pcolormesh(xx, yy, xx**2 + 4*yy**2, shading='gouraud')
fig.show()

meshMtd = QuadMeshMtd(mesh._meshWidth, mesh._meshHeight, mesh._coordinates)
meshMtd.update_from(mesh)
```

DavidFair, 2022-02-03:
Me and Richard discussing (Ab)using Python duck typing / a decorator pattern to avoid copying dict.
We don't directly instantiate a QuadMesh so we must wrap the instance instead
```python
from unittest import mock

class QuadMesh:
    def __init__(self):
        self.foo = False

    def get_x_array(self):
        return 1

class Wrapper:
    def __init__(self, wrapped: QuadMesh):
        self._wrapped = wrapped

    def __getattr__(self, item):
        return getattr(self._wrapped, item)

    def get_array(self):
        return 2 + self._wrapped.get_x_array()

    def not_in_parent(self):
        return False

instance = Wrapper(QuadMesh())
print(instance.get_array())
print(instance.not_in_parent())

mocked = mock.create_autospec(instance)
print(mocked.get_array())
mocked.bar()
```

PR #33509 (2022-02-09): Fix non-ortho view clim autoscaling for MDHisto workspaces in sliceviewer
**Description of work.**

Use decorator/wrapper for QuadMesh with get_array_clipped method. Autoscaling of clim requires image object (in this case QuadMesh output from pcolormesh) to have method `get_array_clipped_to_bounds` (as in `SamplingImage` and `ModestImage`).

**To test:**

(1) Make a event workspace
```
md_3D = CreateMDWorkspace(Dimensions=3, Extents=[-1,1,-2,2,-0.1,0.1], 
    Names="H,K,L", Frames='HKL,HKL,HKL',Units='r.l.u.,r.l.u.,r.l.u.')
FakeMDEventData(InputWorkspace=md_3D, UniformParams='1e4') # 4D data
e [... cut]
closed 2022-02-09 by martyngigg