mantidproject/mantid #33434: Sliceviewer doesn't close when delete original workspace of MDHisto

RichardWaiteSTFC, 2022-02-02:
**Describe the bug**
Sliceviewer doesn't close when delete original workspace of MDHisto - it should because the support for dynamic rebinning has changed

**To Reproduce**
(1) make the workspaces
```
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[-2,2,-1,1,-1.5,1.5,-0.25,0.25], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D, UniformParams='5e5') 
BinMD(InputWorkspace='md_4D', AxisAligned=False, BasisVector0='H,r.l.u.,1.0,0.0,0.0,0.0', BasisVector1='K,r.l.u.,0.0,1.0,0.0,0.0', BasisVector2='L,r.l.u.,0.0,0.0,1.0,0.0', BasisVector3='E,meV,0.0,0.0,0.0,1.0', OutputExtents='-2,2,-1,1,-1.535,-1.435,-0.297,-0.197', OutputBins='100,100,1,1', NormalizeBasisVectors=False, OutputWorkspace='md_4D_svrebinned')
```
(2) Open `md_4D_svrebinned` in sliceviewer
(3) Delete `md_4D` in the ADS

**Expected behavior**
It should close the sliceviewer window and print warning

RichardWaiteSTFC, 2022-02-02:
@martyngigg - we can easily put a check in the ADS for the original workspace existing here
https://github.com/mantidproject/mantid/blob/4cca05c35a35c2abaae7a551b5ddb3956a3001d3/qt/python/mantidqt/mantidqt/widgets/sliceviewer/model.py#L126
is it worth doing that for the release? It won't fix the underlying memory leak of course.

martyngigg, 2022-02-03:
@RichardWaiteSTFC Thanks for highlighting this. Is this a regression?

I'm worried that the code around closing/deletion is quite tricky to reliably know you have it correct so at this stage then I think it might be too late for 6.3. I would be happy for the fix to be worked on now and go to 6.4 though.

RichardWaiteSTFC, 2022-02-03:
> md_4D = CreateMDWorkspace(Dimensions=4, Extents=[-2,2,-1,1,-1.5,1.5,-0.25,0.25], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV') [...]

@martyngigg - just checked on 6.2.0 and it behaves the same so not a regression so will keep milestone at 6.4 S1