mantidproject/mantid #33436: Sliceviewer zoomed in outside of data with log scaling

DanielMurphy22, 2022-02-02:
**Describe the bug**
For Matrixworkspace (event and 2D) and MDHistoworkspaces, you get an errorreporter if you perform these steps in any order:
- Zoom out then in outside the data
- Change the colorbar scaling to Log

I think this problem doesn't exist for MDEventWorkspaces as the dynamic rebinning helps and pretends this out of bounds area is valid data with 0 counts.

**To Reproduce**
- Load CNCS_7860.nxs from TrainingCourseData
- Open sliceviewer by rightclicking in the workspace and selelcting show slice viewer
- Change the colorscale to Log
- Zoom out then in on a region outside of the extent of the data
- Error reporter and this error

**Expected behavior**
Colorbar max and min both set "to 0" and no errorreporter. Possibly also change automatically from Log -> Linear and disallow Log.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS:Ubuntu 18.04 but similar behaviour on Windows 10 (may need to perform the 2 key steps in a certain order on Windows)
 - Mantid Version release candidate 6.3

gemmaguest, 2022-03-11:
I've seen the same error with MAR11060, Log scale, when zooming in on an area with no counts. `values must all be positive` appears as an error in the log, then when I click Home it appears as an unhandled exception.

closed 2022-09-16 by stale[bot]

DanielMurphy22, 2022-09-16:
This is fixed! NOt sure which PR had the fix, but now the warning `Attempted to set non-positive bottom ylim on a log-scaled axis.
Invalid limit will be ignored.` is printed to the Messages Box and the sliceviewer continues gracefully while setting the colorbar scale to linear.