mantidproject/mantid #33437: Update Instrument Viewer documentation

DannyHindson, 2022-02-03:
The documentation for the Instrument Viewer that is linked from the UI is: https://www.mantidproject.org/Instrument_View

I think this is a wiki page rather than in GitHub and it's out of date in several ways:
- it refers to MantidPlot in the title
- some of the screenshots are out of date and are missing buttons or menu options (eg Pick tab - draw toosl for circular sector, arbitrary shape and on bottom row peak comparison and new whole instrument buttons)
- some of the features\buttons don't have descriptions

There's also an rst file here: docs\source\workbench\instrumentviewer.rst. Perhaps this is supposed to be the main one now?

This task is to straighten out which is the main documentation page and to update the documentation. Also link that update page to the instrument viewer ? button in the UI

martyngigg, 2022-05-05:
I'm going to reassign this as a bug. As out of date docs are not a good look either. I'll set the milestone to 6.4 as I think this is worth fixing for that.

mantid-builder, 2022-06-10:
Fixed by #34039

PR #34039 (2022-06-10): Fix Help Information in the Instrument Viewer
**Description of work.**
Directed the instrument viewer to the correct documentation (for workbench, rather than plot) and have it open in the qthelp viewer rather than a web page.

Also updated the `.rst` documentation to reflect the actual functionality present in the GUI. 

**To test:**
1. Delete `doctrees`
2. Build the `docs-qthelp` target. 
3. Build and launch workbench
4. Open the instrument viewer on a workspace
5. Click the ``?`` button
6. The workbench help should appear with the new docs.
7. Check the docs for formatting, spelling, and gr [... cut]
closed 2022-06-10 by mantid-builder