mantidproject/mantid #33478: Windows Fitting Smoke Tests

DanielMurphy22, 2022-02-08:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

RichardWaiteSTFC, 2022-02-08:
[image]
Changing value of a parameter in fit browser doesn't change the value of any tied parameters (though the guess is updated correctly tied parameter in fitbrowser

AnthonyLim23, 2022-02-08:
If you add a user function then press fit or show guess, an error will appear in the logger. The next time you press either button Mantid crashes https://github.com/mantidproject/mantid/issues/33505

RichardWaiteSTFC, 2022-02-08:
Can reproduce this from 6.2 smoke testing (low priority) 
https://github.com/mantidproject/mantid/issues/32605#issuecomment-927917913

RichardWaiteSTFC, 2022-02-08:
Can reproduce this issue (also found during 6.2 smoke testing).
#31660
I think as the comment from @thomashampson in the issue says, the real issue is that the value of the tie isn't updating when the user changes the parameter value

RichardWaiteSTFC, 2022-02-08:
Can reproduce this issue from v6.2 smoke testing as well
#31662

RichardWaiteSTFC, 2022-02-08:
Have found nothing release breaking (no regressions)

closed 2022-02-08 by RichardWaiteSTFC